verilator/verilator #2568: Tristate does not propagate in Altera's BRAM

mjc0608, 2020-09-24:
Verilator version: Verilator 4.101 devel rev v4.100-22-g6430743b (git master)

The problem occurs when simulating an IP from Altera's `simlib` called `altsyncram` (located in `220model.v` and `altera_mf.v`, which I'm not sure whether is ok to upload). When using the following top module and testbench, the wire `wen` does not propagate to `altsyncram_body.wren_a` (i.e., I set `wen` to 1, but `wren_a` is still 0).

``` verilog
module testram(
    input logic clk,
    input logic wen,
    input logic [3:0] waddr,
    input logic [15:0] wdata,
    input logic [3:0] raddr,
    output logic [15:0] rdata
);

    altsyncram #(
        .intended_device_family("Stratix"),
        .operation_mode("DUAL_PORT"),
        .width_a(16),
        .widthad_a(4),
        .numwords_a(16),
        .width_b(16),
        .widthad_b(4),
        .numwords_b(16),
        .rdcontrol_reg_b("CLOCK0"),
        .address_reg_b("CLOCK0"),
        .outdata_reg_b("UNREGISTERED"),
        .read_during_write_mode_mixed_ports("OLD_DATA")
    ) bram (
        .clock0(clk),

        .wren_a(wen),
        .address_a(waddr),
        .data_a(wdata),
        .rden_a(1'b0),

        .address_b(raddr),
        .q_b(rdata),
        .wren_b(1'b0),
        .rden_b(1'b1),

        .data_b(),
        .clock1(),
        .clocken0(),
        .clocken1(),
        .clocken2(),
        .clocken3(),
        .aclr0(),
        .aclr1(),
        .byteena_a(),
        .byteena_b(),
        .addressstall_a(),
        .addressstall_b(),
        .q_a(),
        .eccstatus()
    );
endmodule
```
``` c++
#include <stdlib.h>
#include <iostream>
#include <stdio.h>
#include <cassert>
#include <verilated_fst_c.h>
#include <verilated.h>

#include "Vtestram.h"

using namespace std;

const uint64_t psc = 2500; // 2500 ps per cycle
const uint64_t pst = 1250; // 1250 ps per tick (half cycle)
uint64_t timestamp = 0;
double sc_time_stamp() {
    return timestamp;
}
void sc_time_step() {
    timestamp += pst;
}

int main(int argc, char **argv) {
    Verilated::commandArgs(argc, argv);
    Vtestram *tb = new Vtestram;

    Verilated::traceEverOn(true);
    VerilatedFstC *trace = new VerilatedFstC;
    tb->trace(trace, 99);
    trace->open("testram.fst");

    tb->wen = 1;
    tb->waddr = 0;
    tb->wdata = 0xff;
    tb->raddr = 0;

    for (int i = 0; i < 10; i++) {
        tb->eval();
        tb->clk = 1;
        tb->eval();
        trace->dump(timestamp);
        sc_time_step();

        tb->clk = 0;
        tb->eval();
        trace->dump(timestamp);
        sc_time_step();
    }

    trace->flush();
    trace->close();

    exit(EXIT_SUCCESS);
}
```
Verilog files are compiled with the following parameters, where `sources.txt` includes `220model.v`, `altera_mf.v`, and `testram.sv`.
``` bash
verilator -trace-max-array 65536 -trace-max-width 65536 -cc -timescale-override 1ps/1ps -Wno-WIDTH -Wno-LITENDIAN -Wno-UNPACKED -Wno-BLKANDNBLK -Wno-CASEINCOMPLETE -Wno-CASEX -Wno-TIMESCALEMOD -Wno-PINMISSING -Wno-COMBDLY -Wno-UNOPTFLAT -Wno-MULTIDRIVEN -trace-fst -trace-structs -F sources.txt -top-module testram
```
It seems to be a problem around how verilator handles `tri0` and `tri1`, because `wren_a` is typed as `tri0`. After changing all `tri0` and `tri1` to `logic` with [this script](https://github.com/twosigma/verilator_support), the signal `wen` can propagate correctly but there are still some other errors (since some weird logic in `altsyncram` depends on `tri0` and `tri1`).

According to [verilator's manual](https://www.veripool.org/wiki/verilator/Manual-verilator#Tri-Inout):
> Verilator converts some simple tristate structures into two state. Pullup, pulldown, bufif0, bufif1, notif0, notif1, pmos, nmos, tri0 and tri1 are also supported. Simple comparisons with === 1'bz are also supported.

I believe this is actually supported and is a bug.

wsnyder, 2020-10-01:
There probably hasn't been much testing of tri0/1 so this is easy to believe has a bug.  Can you please try to make a test case that doesn't require the altera code?

mjc0608, 2020-10-05:
Can you try the following? I noticed that l2.clock is always 1 and l2.wren is always 0.

``` verilog
module test(
    input logic clk,
    input logic wen
);

    l1 l1_inst(
        .wren(wen),
        .clock(clk)
    );

endmodule

module l1 (
    wren,
    clock
);

    input  wren;
    input  clock;

    tri0 wren;
    tri1 clock;

    l2 l2_inst(
        .wren(wren),
        .clock(clock)
    );

endmodule
 
module l2 (
    wren,
    clock
);

    input wren;
    input clock;

    tri0 wren;
    tri1 clock;

endmodule
```
``` c
#include <stdlib.h>
#include <iostream>
#include <stdio.h>
#include <cassert>
#include <verilated_fst_c.h>
#include <verilated.h>
#include <queue>
#include <map>
#include <tuple>

#include "Vtest.h"

using namespace std;

const uint64_t psc = 2500; // 2500 ps per cycle
const uint64_t pst = 1250; // 1250 ps per tick (half cycle)
uint64_t timestamp = 0;
double sc_time_stamp() {
    return timestamp;
}
void sc_time_step() {
    timestamp += pst;
}

int main(int argc, char **argv) {
    Verilated::commandArgs(argc, argv);
    Vtest *tb = new Vtest;

    Verilated::traceEverOn(true);
    VerilatedFstC *trace = new VerilatedFstC;
    tb->trace(trace, 99);
    trace->open("test.fst");

    tb->wen = 1;

    for (int i = 0; i < 10; i++) {
        tb->eval();
        tb->clk = 1;
        tb->eval();
        trace->dump(timestamp);
        sc_time_step();

        tb->clk = 0;
        tb->eval();
        trace->dump(timestamp);
        sc_time_step();
    }

    trace->flush();
    trace->close();

    exit(EXIT_SUCCESS);
}
```

wsnyder, 2020-11-06:
This is effectively a duplicate of #1373.  Verilator doesn't support top level tristates, so basically it doesn't know if the input port (which got promoted to inout due to attachments) is being driven or not.  Currently it assumes not and doesn't know otherwise, so the tristate (to Verilator) net then falls to the default "0" or "1" for tri1.

It's probably more logical for Verilator to assume that the input is always being driven. This would break previos behavior so a flag would be needed, if someone would like to contribute this it would be useful.  The real fix as described in #1373 is for Verilator to understand tristate inputs.

You should be able to work around this by adding a buf between the top level input and lower module.

wsnyder, 2020-11-06:
BTW sorry for the delayed response, normally try to get back much sooner but somehow this got missed.

mjc0608, 2020-11-06:
Thanks! And I'm going to close the issue.

closed 2020-11-06 by mjc0608