verilator/verilator #2570: Error-BLKLOOPINIT for unroll count > 8192

rayx999, 2020-09-25:
I got 2 BLKLOOPINIT errors:

%Error-BLKLOOPINIT: spsb_Svt_1024x32m4_R.v:3637:28: Unsupported: Delayed assignment to array inside for loops (non-delayed is ok - see docs)
 : ... In instance genisys.mdio_master_i0.mdio_master_mac_i0.mem_wrpr_i0
 3637 |       u_ram_core.memory[i] <= #0.005 {33{1'bx}};
      |                            ^~
%Error-BLKLOOPINIT: hdspmb_SegQ_Svt_8192x21m8_R.v:3330:28: Unsupported: Delayed assignment to array inside for loops (non-delayed is ok - see docs)
                                                                                                                                                                                         : ... In instance genisys.plug_if_top_i0.mdio_slave_mwisfq_i0.mdio_slave_mac_i0.mem_wrpr_i0.mem_tech_lib_i0.mdio_slave_mem_21b_ram_sp_65536x21.mdio_slave_mem_21b_ram_sp_65536x21_i0.ROW[0].COL[0].tsmc_hdspmb_SegQ_Svt_8192x21m8_R_wrapper__i
 3330 |       u_ram_core.memory[i] <= #0.005 {22{1'bx}};

When I add --unroll-count 10240, the first error disaapears, second one persists. The loop count in that code is 8192:

//  Flush all memory content
task xMemoryAll;
  integer i;
  begin
    for (i = 0; i < 8192; i = i + 1) begin
      **u_ram_core.memory[i] <= #0.005 {22{1'bx}}**;
    end
    if (f_print_info) $display ("\tINFO %m : memory is flushed at %t", $realtime);
  end
endtask

--unroll-stmts 8192 doesn't work.

Any other way to get over this error?

Regards,
Ray

wsnyder, 2020-10-01:
You can either change the <= to just =, or comment out that line since it's presumably only doing this when something illegal is done to the memory.

Not sure why the --unrol-count 10240 wouldn't work, it should.  However this will make the code slower than the two alternatives above.l

closed 2020-10-01 by wsnyder

Owen199x, 2023-11-09:
you should update the new value with larger size number of unroll-stmts.