verilator/verilator #2574: Assignment statement does not take effect.

yanx21, 2020-09-28:
Hi, Show you my testcase directly.

```verilog
// demo.v
module t (
    input   logic   clk,
    output  logic   en_pp
);
    sub U_SUB (
        .clk        (clk),
        .en         (1'b1),
        .en_pp      (en_pp)
    );
endmodule

module sub (
    input   logic   clk,
    input   logic   en,
    output  logic   en_pp
);
logic pp[1:0];
always @ (*) begin
    pp[0] = en;
    en_pp = pp[1]; // This assignment statement does not take effect.
end
always @ (posedge clk) begin
    pp[1] <= pp[0];
    $display("en_pp: %d", en_pp);
end
endmodule
```
```cpp
// sim_main.cpp
#include "Vdemo.h"
#include "verilated.h"

vluint64_t main_time = 0;
double sc_time_stamp() { return main_time; }

int main(int argc, char** argv, char** env) {
    Verilated::commandArgs(argc, argv);
    Vdemo* top = new Vdemo;
    while (!Verilated::gotFinish() && main_time < 40) {
        top->eval();
        main_time++;
        top->clk = !top->clk;
   }
    delete top;
    exit(0);
}
```
0 error or warning when translating and running the case.
```txt
// Run the testcase and output as bellow.
en_pp: 0
en_pp: 0
en_pp: 0
en_pp: 0
en_pp: 0
en_pp: 0
en_pp: 0
en_pp: 0
en_pp: 0
en_pp: 0
en_pp: 0
en_pp: 0
en_pp: 0
en_pp: 0
en_pp: 0
en_pp: 0
en_pp: 0
en_pp: 0
en_pp: 0
en_pp: 0
```

As you can see, en_pp were all 0, which was absolutely incorrect. It should be 1 after cycle 1.
And I looked into the cpp code, I could find the code related to the assignment `en_pp = pp[1]` in Vdemo__Slow.cpp
```cpp
void Vdemo::_settle__TOP__2(Vdemo__Syms* __restrict vlSymsp) {
    VL_DEBUG_IF(VL_DBG_MSGF("+    Vdemo::_settle__TOP__2\n"); );
    Vdemo* const __restrict vlTOPp VL_ATTR_UNUSED = vlSymsp->TOPp;
    // Body
    vlTOPp->t__DOT__U_SUB__DOT__pp[0U] = 1U;
    vlTOPp->en_pp = vlTOPp->t__DOT__U_SUB__DOT__pp[1U]; // code related to assignment.
}
```
But I failed to find any code related to the assignment `en_pp = pp[1]` in _combo_ function or _sequent_ function.
What's more, I studied some source code, maybe the assignment had been deleted in V3Order by marking it as delete domain?

What's your opinion?

Thanks.

wsnyder, 2020-09-28:
This should have thrown a BLKANDNBLK error, as the scheduler doesn't support mixed array assignments like this.  If you use this

    logic [1:0] pp;

it will.  One workaround is to use

    logic pp /*verilator split_var*/;

 It's certainly a bug that this isn't an error (and of course a known long term bug to support this).

gezalore, 2026-11-01:
Simulates correctly on master with no warning.

closed 2026-11-01 by gezalore