verilator/verilator #2576: Continuously assign a value to the same signal.

yanx21, 2020-09-29:
Hi,

I got a testcast whose behavior was not my expectation.

```verilog
module t (
    input   logic     clk,
    input   logic     mask,
    output  logic     dout
);
logic      d0;
logic      dpp[1:0];
always @ (posedge clk) begin
    d0 <= ~d0;
end
always @ (*) begin
    dpp[0] = d0;
    dout   = dpp[1];
end
always @ (*) begin
    dpp[1] = 0; // This is the key statement.
    dpp[1] = dpp[0] & mask;
end
endmodule
```
```cpp
// sim_main.cpp
#include "Vdemo.h"
#include "verilated.h"
vluint64_t main_time = 0;
double sc_time_stamp() { return main_time; }
int main(int argc, char** argv, char** env) {
    Verilated::commandArgs(argc, argv);
    Vdemo* top = new Vdemo;
    top->mask = 0x1;
    printf("  clk  dpp[0]  dpp[1]\n");
    while (!Verilated::gotFinish() && main_time < 20) {
        top->eval();
        printf("   %d     %d      %d\n",
            top->clk,
            top->t__DOT__dpp[0],
            top->t__DOT__dpp[1]);
        main_time++;
        top->clk = !top->clk;
    }
    delete top;
    exit(0);
}
```
```txt
// screen output
  clk  dpp[0]  dpp[1]
   0     0      0
   1     1      0
   0     1      1
   1     0      1
   0     0      0
   1     1      0
   0     1      1
   1     0      1
   0     0      0
   1     1      0
   0     1      1
   1     0      1
   0     0      0
   1     1      0
   0     1      1
   1     0      1
   0     0      0
   1     1      0
   0     1      1
   1     0      1
```
The result shows that `dpp[0]` toggles at posedge of `clk` and  `dpp[1]` toggles at negedge of `clk`, but they should change simultaneously.

I tried to study the source code to find out the reason, and then, I found the the comments in the source code like this:
https://github.com/verilator/verilator/blob/222ca4b009baec47f04e99d32373b3a1fd440d05/src/V3Order.cpp#L1043-L1055

May I ask why you make this assumption? It seems you are doing some optimization by reducing the scale of the graph, but It may cause some problem.
If this is your deliberate decision, I think you'd better give some warning here.

Thanks.

Best regards.

wsnyder, 2020-09-29:
I'm impressed you were able to trace that down into the ordering code.

As you probably figured out the always dpp[0] block should be evaluating multiple times per eval (and give a UNOPTFLAT warning), but Verilator thinks it is safe not to do so.

This is a bug, basically the description in V3Order should in most cases be correct for a single variable ("foo"), but would be incorrect for foo[1] which is your case.  So this bug is a combination of that code, and that Verilator at present does ordering and most optimizations on an entire variable ("dpp") rather than vector or bit by bit ("dpp[0]" vs "dpp[1]").

BTW you're hitting these cases because of the assignments to the same variable in different always blocks, which is an unusual style, and #2569 will soon warn about this (as a style warning but didn't think until now it might have to be a it-will-simulate wrong warning).

This code is also fixed with

     logic dpp[1:0] /*verilator_split_var*/

as then dpp[0] and dpp[1] are scheduled separately.  

Long term fix is to schedule bits separately.

riteme, 2021-06-25:
I'm experiencing an issue identical to this one and #2673 on Verilator v4.200. `/* verilator split_var */` does help. I can provide a shorter testcase for this issue:

```verilog
// Echo.sv
module Echo (
    input logic in_value,
    output logic out_value
);
    struct packed {
        logic left, right;
    } pipe;

    assign pipe.left = in_value;
    assign out_value = pipe.right;

    always_comb begin
        pipe.right = 1'b0;  // (*)
        pipe.right = pipe.left;
    end
endmodule
```

```c++
// Echo.cpp
#include <cstdint>
#include "obj_dir/VEcho.h"
#include "verilated.cpp"

int main() {
    auto top = new VEcho;

    top->in_value = 1;
    top->eval();
    printf("%d\n", top->out_value);

    return 0;
}
```

Expected output is `1`.

As mentioned above, the line marked with `(*)` will suppress `UNOPTFLAT` warning and lead to incorrect result.

rapiz1, 2023-02-21:
I may hit on this with my packed struct for AXI bus. The behaviour I observed is like this:
```
assign x = y; // x is a member in packed struct, y is a member in interface
```
And y changes on posedge, x changes on negedge, just half the clock cycle after y.
[image]

Adding `/* verilator split_var */` does not help.

I use definitions and assignments from [pulp-platform/axi](https://github.com/pulp-platform/axi/blob/fd60be8b51a4fa7476856be162ce3334474592ba/include/axi/assign.svh#L26)

And more snippets if they're useful:
```
module foo (AXI_BUS.Master m_axi);
//...
  `AXI_TYPEDEF_ALL(axi, addr_t, id_t, data_t, strb_t, user_t)
  axi_req_t mmu_s_axi_req /*verilator split_var*/;
  axi_req_t mmu_m_axi_req /*verilator split_var*/;
  axi_resp_t mmu_s_axi_resp /*verilator split_var*/;
  axi_resp_t mmu_m_axi_resp /*verilator split_var*/;

  `AXI_ASSIGN_FROM_REQ(m_axi, mmu_m_axi_req);
  `AXI_ASSIGN_TO_RESP(mmu_m_axi_resp, m_axi);

  `AXI_ASSIGN_TO_REQ(mmu_s_axi_req, mux_m_axi);
  `AXI_ASSIGN_FROM_RESP(mux_m_axi, mmu_s_axi_resp);
//...
```
The macro expands to assignments to `logic`.

Sorry that I failed to reduce it into a minimal case.  I don't know what to do at this moment. Is `/*verilator split_var*/` on the declaration sufficient? Do I need to put it elsewhere like these types and members are defined?

wsnyder, 2023-02-23:
Likely a split_var is needed on every variable used as an assignment.  If it is propagated to an interface that does an assignment, then the whole interface is assigned I'm not sure if the split_var will do what you want at all, some experimentation is needed.

daglem, 2023-04-29:
Possibly related, I'm having problems getting non-blocking assignments to struct members working.

Something like this yields `a = 0`:
```
always_ff @(posedge clk) begin
    a <= 0;
    if (something)
       a.b <= 1;
    else
       a.b <= 2;
end
```

Tagging `a` with `/*verilator split_var*/` doesn't make any difference.

Is this a known issue?

wsnyder, 2025-07-03:
Verilator for a while has split the variables automatically, and this test passes.

closed 2025-07-03 by wsnyder