verilator/verilator #2577: Help needed on split RTL design vs testbench/test SystemC/C++ compile, and link to run.

ayewinoung, 2020-09-30:
I'm trying to compile all the RTL design files into lib (top_module.a) and compile the rest of the testbench & test, SystemC/C++ files in different compile and link against the "top_module.a".  All RTL design files compile and generate "top_module.a"
I compile the "verilated.cpp", "verilated_dpi.cpp" & "verilated_threads.cpp" together with the testbench/test SystemC/C++ files.  However, when I try to compile, I got the following error.

In file included from /usr/local/share/verilator/include/verilated_threads.cpp:18:
/usr/local/share/verilator/include/verilated_threads.h:48:17: error: no template named 'atomic' in namespace 'std'
    static std::atomic<vluint64_t> s_yields;  // Statistics

I feel like, I'm fundamentally missing something here, there may be some hidden issues in link step.
And I'm not good with "Makefiles" and try to understand as much as I can on multiple makefile structure inside examples.
Btw, I've existing design which compile & run fine, if I compile all of the RTL design files, testbenh SystemC & tests C++ files in one giant compile step.
As always thanks in advance.

wsnyder, 2020-09-30:
verilated_threads.cpp should only be compiled with multithreaded models (verilator with --threads).  Alternatively you can pass -DVL_THREADED to the compile-time to always be able to compile this header.  I'll add a more obvious error.

wsnyder, 2020-09-30:
BTW this error means "<atomic>" was not included.

Added comment.

Closing but feel free to ask additional related questions on this thread if needed - if it's a new problem requiring a fix please open a new ticket. Thanks.

closed 2020-09-30 by wsnyder