verilator/verilator #2578: Compiling big design issue(s)...

ayewinoung, 2020-10-02:
I'm trying to compile a relatively big design in flat mode (i.e. not --hierarchical), and run into following "clang" compiler failure with "verialted" cpp files. It finished verliating, and failed at C++ compile step.

sirius_tb__2.cpp:5:10: fatal error: sorry, this include generates a translation unit too large for Clang to process.
#include "sirius_tb.h"

"sirius_tb" is top level RTL module name. 
The design took ~ 17hrs and 470Gbyte of memory, with latest EPYC 7571  to verliate it.  And generate ~ 10Gbyte of cpp files ! And shame couldn't compile it ;)

Verilator 4.101 devel rev v4.100-26-gf8ab30d1
Ubuntu clang version 11.0.0-++20200928063537+eb83b551d3e-1~exp1~20200928164152.107

I'll be happy to provide more details, if its helpful. 

FYI - when used ( "with" --hierarchical) design will compile in respectable 4min, 9Gbyte memory.  However, it can only be run at single threaded mode (slow sim time). 

Any thoughts/comments appreciated.

wsnyder, 2020-10-02:
How large is the file (and headers)?  You could use clang preprocess only (-E) to get one big output, then you can see the total size.

Did you try GCC?  (Fortunately you can try this without reverilating, just change CXX in your environment).  Note gcc -Os is typically within 10% of clang.

Are you running the compile on the same huge machine to make sure isn't out of memory?

Once you have these answers you may want to post to the clang bug forums to see if there is a switch or something you can do.  Also if we can understand what the limit is perhaps we can change the code emitter to avoid the restriction.

I'm amazed hierarchical was so much faster!

ayewinoung, 2020-10-03:
Thanks for looking into it, will get back with details..

wsnyder, 2020-11-07:
Still awaiting more information.

closed 2020-12-11 by wsnyder

fskoras, 2022-02-26:
I'm having simillar problem in my custom static analyzer tool running on top of libclang. It looks like there is a huge desig flaw in the SourceManager that developer don't want to fix for some reason. Like stated in the source commnet. "Check that the file's size fits in an 'unsigned' (with room for a past-the-end value). This is deeply regrettable, but various parts of Clang (including elsewhere in this file!) use 'unsigned' to represent file offsets, line numbers, string literal lengths, and so on, and fail miserably on large source files."

You can find more info in the link below:
https://groups.google.com/g/llvm-dev/c/WnbtHnb0wtA