verilator/verilator #2579: Verilator should 'clean' inputs in CC mode

stephenry, 2020-10-02:
Verilator 4.101 devel rev v4.100-27-g5033c906

I have been working on a little project at: https://github.com/stephenry/m; and have encountered an obscure problem. Naturally I immediately suspected my dodgy RTL, but it turns out to relate to the means by which state is communicated between my C++ to the Verilated C++.

In the test-bench, we see the following code:

```systemverilog
  m_pkg::in_t                      in_w;

  always_comb begin : in_PROC

    in_w                       = '0;
    in_w.sop                   = in_sop_w;
    in_w.eop                   = in_eop_w;
    in_w.length               = in_length_w;
    in_w.data                  = in_data_w;
```

Where m_pkg::in_t is defined as:

```systemverilog
  // Input packet type
  typedef struct packed {
    logic        sop;
    logic        eop;
    len_t        length;
    data_t       data;
  } in_t;
```

Nothing too controversial. The generated C++ for the in_PROC, however, is:

```cpp
    vlTOPp->tb__DOT__in_w[2U] = ((0xfU & vlTOPp->tb__DOT__in_w[2U]) 
                                 | (0xfffffff0U & ((IData)(vlTOPp->in_sop_w) 
                                                   << 4U)));
    vlTOPp->tb__DOT__in_w[2U] = ((0x17U & vlTOPp->tb__DOT__in_w[2U]) 
                                 | (0xfffffff8U & ((IData)(vlTOPp->in_eop_w) 
                                                   << 3U)));
    vlTOPp->tb__DOT__in_w[2U] = ((0x18U & vlTOPp->tb__DOT__in_w[2U]) 
                                 | (IData)(vlTOPp->in_length_w));
```
The code sets `SOP`, which is located at [4]; set `EOP` which is located at bit [3]; and then sets length with is the field [2:0].

The length field is considered only on the `EOP`. In my test environment, in the `EOP` == 'b0 case, `LENGTH` was being set inadvertently to some unknown and uninitialized value that I didn't care about.

As the generated C++ does not explicitly constrain the input such that it contains values only within the correct bit-range, any values outside of this range (essentially bits [7:3] of the `LENGTH` vluint8_t) can potentially set `SOP` and `EOP`. In my specific test case, `in_w.eop` was begin set when `in_eop_w` was 'b0, but when length (which was being otherwise ignored) had `'8b????_1???`. The test would then fail in a non-deterministic manner and, when failing, the waveform would indicate unusual and incorrect results.

I suspect Verilator may wish to additionally mask inputs such that only bits within the valid range are considered.

wsnyder, 2020-10-02:
Verilator does what you suggest, but you need "-DVL_DEBUG" defined, as it only does this under debug compiles for speed reasons.

wsnyder, 2020-10-02:
%Error: unknown:0: Testbench C set input 'clk' to value that overflows what the signal's width can fit

closed 2020-10-02 by wsnyder