verilator/verilator #2582: error: #error "Verilator requires a C++11 or newer compiler"

adumont, 2020-10-04:
After upgrading to v4.100, when I build my project (a Qt GUI which includes a Verilator model), I get the following error:

error: #error "Verilator requires a C++11 or newer compiler"

The complaining command is:

```
g++ -c -m64 -pipe -O2 -Wall -W -D_REENTRANT -fPIC -DQT_NO_DEBUG -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_CORE_LIB -I. -I../verilog/obj_dir -I/home/adumont/toolchain/share/verilator/include -isystem /usr/include/x86_64-linux-gnu/qt5 -isystem /usr/include/x86_64-linux-gnu/qt5/QtWidgets -isystem /usr/include/x86_64-linux-gnu/qt5/QtGui -isystem /usr/include/x86_64-linux-gnu/qt5/QtCore -I. -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++-64 -o main.o main.cpp

In file included from /home/adumont/toolchain/share/verilator/include/verilated.h:27:0,
                 from /home/adumont/toolchain/share/verilator/include/verilated_heavy.h:27,
                 from ../verilog/obj_dir/Vtop.h:11,
                 from mainwindow.h:8,
                 from main.cpp:1:
/home/adumont/toolchain/share/verilator/include/verilatedos.h:198:3: error: #error "Verilator requires a C++11 or newer compiler"
 # error "Verilator requires a C++11 or newer compiler"
```

Should I add any flag/environ variable to the build?

Here's my g++ version:

```
$ g++ -v
Using built-in specs.
COLLECT_GCC=g++
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/5/lto-wrapper
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 5.4.0-6ubuntu1~16.04.12' --with-bugurl=file:///usr/share/doc/gcc-5/README.Bugs --enable-languages=c,ada,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-5 --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-libmpx --enable-plugin --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-5-amd64/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-5-amd64 --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-5-amd64 --with-arch-directory=amd64 --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --enable-objc-gc --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu1~16.04.12)
```

I build verilator from source. When I rollback verilator to bdecf6c4 (tag: v4.040), my project builds fine again, so I guess it's due to something introduced between v4.040 and v4.100.

(My project is here: https://github.com/adumont/ad6502).

adumont, 2020-10-04:
I've found I need to for C++11 in gcc , and I can enable it from Qt project file by adding:

`CONFIG += c++11
`
That fixes the build with v4.100 :+1: 

Source: https://stackoverflow.com/a/23615295/728281

closed 2020-10-04 by adumont

wsnyder, 2020-10-04:
Yes, that makes sense, thanks for self-solving.