verilator/verilator #2583: Improve --hierarchical performance

yTakatsukasa, 2020-10-04:
#1520 also improves hierarchical verilation.

Multi-threaded run (`--threads n`)  is not as effective as flat verilation found in  #2562 .

Possible ideas are
- Annotate the simulation cost of hier_block via protect-lib wrapper so that upper module can schedule properly
  - Upper modules only see the sv wrapper of the protect-lib, so its simulation cost looks much cheaper than the actual hier_block
- Share the thread pool among hier_blocks
  - Currently hier_blocks are verilated with `--threads 1` to avoid overcommit

Needs new ideas and experiments.

ayewinoung, 2020-10-11:
Thanks for looking into this.
IMO: If we can scale across multiple threads relatively well, with "--hierarchical", *this* will be the killer *feature* that even big 3 simulators don't do well with very large designs.  
Here is my thoughts, and for whats its worth 

- In future, and most of new silicon designs now days, are implementation of the same "module" multiple times (e.g. AI engines/cores, graphics, CPU, hyper scale processing, etc.)
- Silicon gate density will keep growing and best use of additional silicon growth is scaling existing "modules", hence, support above point
- In addition to existing fine grain multi-threading, if there is option in verilator to assign a run time thread per "--hierarchical" module, it will scale very well for designs with multiple instances of  "--hierarchical" module.

Full disclosure, I've design which instantiate the same module 1000 time and simulations times are taking days, and could do with effective multi threaded scaling.
I'm happy to do experiments and report back on things, if that helps.  
Also will post more in depth study I did on compile time/peak mem usage and run time/peak mem usage, on multiple designs sizes with verilator, flat vs hierarchical  and one of the big 3.

God speed ;)

wsnyder, 2020-10-12:
I think we're all in agreement that thread sharing is very useful, it's a matter of getting it done ;)

I'd propose the ideal way to fix this is to somehow have global knowledge of the thread partitioning as the hierarchical model is being formed, either

1. statically when creating it, that is the global partitioning somehow determines a good static allocation of threads between the parent and children and tells each appropriately what it can use.  Perhaps a step along this path is for the user to manually recommend the number of threads in the children.

2. dynamically, one possibility is the children are made for maximum (virtual) thread use but then the parent can map to fewer physical threads.  This seems much harder as we can't simply map the existing thread algorithms onto this, as each virtual thread is designed to block.

wsnyder, 2025-01-20:
First item resolved in #5734.

b-chmiel, 2025-01-28:
> I'd propose the ideal way to fix this is to somehow have global knowledge of the thread partitioning as the hierarchical model is being formed, either
>
>     1. statically when creating it, that is [...]

I don't think that sharing thread pool would cut it.

I'd created a PoC that dynamically assigns indexes to the static thread pool and the following problems arose:

1. Wasted thread pool workers. One worker per one hier block instance is always sacrificed, as it waits until all tasks of this instance are finished.
2. Inaccurate scheduling. One cannot guess how many `eval` rounds will be performed inside children, only a single cost is possible to be determined.
3. Multiple `eval` loops. Related to the point above, this might induce performance problems as well.
4. UNOPTFLAT-related sensitivity problems. With this approach, we join sensitivities of possibly unrelated signals.
 

I propose a **different approach**: schedule hier childs' MTasks in parent via DPI-s in a hierarchical ProtectLib wrapper.

MTasks will be exposed as DPI calls and will be invoked from SystemVerilog hierarchical wrapper. This is similar to the current implementation except that multiple `update` functions would be created, one per each MTask.

We would still schedule MTasks in hierarchical children but instead of emitting `Vthread_nba__x` functions with MTask chains, emit them in the hierarchical SystemVerilog wrapper with dependencies and triggering mechanisms modeled there as well. We could also need to expose trigger checks there and react to whole module retriggers.

At first, this might be implemented in a way, that each MTask would be wrapper with a separate `eval` loop in a hier child. After that, we may unify `eval` loops.

With this approach we may also have a chance to tackle UNOPTFLAT warnings (see https://github.com/verilator/verilator/pull/5669#issuecomment-2536100157) by making sensitivities of a separate exposed MTasks explicit and thus preserving minimal dependencies between I/O ports.

@wsnyder @yTakatsukasa What are your thoughts on that?

wsnyder, 2025-01-29:
I'd have two concerns related to mtask calling as you suggest:
1. The DPI routines add overhead, and there's a lot of mtasks, so this will harm performance.
2. Changing the lower modules (that create mtasks) in any way will expose changes to the top instantiating hierarchical model.  Part of the intent of hierarchical is to avoid needing to recompile nearly everything if a low-level design changes.

yTakatsukasa, 2025-02-16:
I'm sorry for my late response.

I think https://github.com/verilator/verilator/issues/2583#issuecomment-2621537989 makes sense.

This is just rough thought: can we have an assumption that may cover popular usecase ?
e.g. if a hierarchical block is instantiated many times, such as recent many core architecture,  may be there is simple but good enough scheduling plan.

closed 2025-03-24 by wsnyder