verilator/verilator #2584: VCD file contains non-ASCII character

simdream, 2020-10-06:
Same small test case worked fine with V4.029, after changed to the following version, VCD can not opened by gtkwave. And noticed some non-ascii code in the VCD file. 

Verilator 4.099 devel rev v4.040-67-gba5779c6

> b10101101011101001011111101000110 4
> �5
> 16
> b11111111110111111111111111111011 7

Anyone has met similar issue with this version?

wsnyder, 2020-10-06:
Can you please create a test to recreate this?  It probably would help to do that to see what signal name is the identifier "5" from the table at the top of the dump.

simdream, 2020-10-08:
found work around.

In testbench.cpp, the code used to written as:
   TOP->clk = ~TOP->clk;
This worked find with V4.029. But in V4.099, the non-ASCII code is 0xff. After we changed the code to :
  TOP->clk = (~TOP->clk) & 0x1;
Now the clk signal in VCD is correct

wsnyder, 2020-10-08:
BTW, If you compile with VL_DEBUG it will warn when you drive upper bits like this, which is illegal.

So looks like working as intended, thanks for debugging.

closed 2020-10-08 by wsnyder