verilator/verilator #2586: Consider a define for FST or generic tracer

markusdd, 2020-10-07:
This is in direct reference to this pull request in cocotb: https://github.com/cocotb/cocotb/pull/2102

Currently there is no way to tell in C++ if FST or VCD tracing was selected on the command line and we 
have to use an ugly -CFLAGS option to pass on another define to make the switch.

a) Is there a nicer option that has been overlooked

b) if not, would you consider adding a VM_TRACE_FST define when --trace-fst is selected, so that you can check for it in the C++ world?

A whole different option would be to not expose the specific tracer type in the top TB, but a a generic tracer class that 'mutates' under the hood depending if no tracing, VCD tracing or FST tracing was selected.

This would completely eliminate the need to check for VM_TRACE in the TB code everytime.
You could always write your code withtracing and could dynamically switch it on and off via CLI
and save the ugly preprocessor macros in the TB code.

wsnyder, 2020-10-07:
This seems reasonable, perhaps you could submit a Verilator pull request to add it, so then you can test with cocotb it does what you're expecting? Thanks.

markusdd, 2020-10-07:
This is also a backward-compatibility question, so what would be preferred?

I suggest setting VM_TRACE as is (meaning for --trace AND --trace-fst) and orthogonally set VM_TRACE_FST as an addition.

Because there might be code out there which checks for the actual value of VM_TRACE=1, so changing it to 2, or vcd or fst might break existing testbenches.

So anyone who wishes to support both can do so, who wants legacy/VCD sticks with VM_TRACE and anyone only interested in FST can just use VM_TRACE_FST.

a) Does that sound reasonable?

b) can you give me a quick guidance in which Makefile script this would need to end up? I did a quick grep over the repo and the way VM_TRACE is being set currently is not totally clear to me, as it's pulled out of a variable, which supposedly is being set by some script/program generating that makefile.

wsnyder, 2020-10-07:
Yes, I think a new define is correct.  It should be defined in the same file as VM_TRACE.

BTW how will this be used?

If to select which trace class perhaps a VM_TRACE_CLASS with the appropriate class is better.

Alternatively verilator now supports $dumpfile etc, perhaps that could be used instead.

markusdd, 2020-10-07:
This pull request gives the example:
https://github.com/cocotb/cocotb/pull/2102/files

markusdd, 2020-10-09:
Also dumpfile is not very nice especially when used with cocotb as you would litter your RTL code with simulation specifics. Also with commercial sims you would not want to use dumpfile, makes the code less portable.
Also dumpfile is not really controllable via CLI, so you always dump waves, even when in batch mode.
Or you have to start using defines again, which brings us back to square one.

markusdd, 2020-10-10:
I prepared a PR for this, thsi shoudl be 100% backward compatible and ease supprtoign multiple trace formats inside one .cpp toplevel

markusdd, 2020-10-11:
Oh, that was quick!

Thanks a lot, will close this then.

closed 2020-10-11 by markusdd