verilator/verilator #2589: Return bool from callValueCbs() similar to callCbs()

garmin-mjames, 2020-10-07:
It can be important to know if any **cbValueChange** callbacks were actually called by `callValueCbs()`, just as `callCbs()` returns.

This is useful when calling `eval()` in a loop along with callbacks, to know when to move to the next time step.

Requested feature:
`callValueCbs()` returns:
- `true` if any registered **cbValueChange** callbacks were called
- `false` if none were called

For context, this is related to ongoing development of https://github.com/cocotb/cocotb support for Verilator (https://github.com/cocotb/cocotb/pull/2105).

wsnyder, 2020-10-07:
Been following the cocotb thread and agree, could you attempt a pull request so this can be tested in context of cocotb before merging? Thanks.

wsnyder, 2020-10-27:
Thanks for your work on this.

PR #2605 (2020-10-27): Return bool from callValueCbs()
Closes #2589.

I'm still working on a test for this.
closed 2020-10-27 by wsnyder