verilator/verilator #2590: Threads 2 gives 2X slower run time performance

simdream, 2020-10-09:
We have an example design which run 2X slower  with --threads 2. the design is almost instnaciate one module two times so supposed to be friendly to multi threading. Checking the generated C code, we do not see any waitUntilUpstreamDone in the two Vmtask__*, hints there is no one thread wait for another in the middle. Any hints on the possible reason? We will also run profile and check, but suspect maybe there are many variables needs to be copied to second thread at the beginning of each _eval_tep()?

simdream, 2020-10-09:
[image]

from 600us - 1250us, parallelism number is 0, does that say the whole run is hold for that time period? 

And what is the unit of the x-axis?

simdream, 2020-10-09:
um... take my word back, looked at wrong database. This one does contains waitUntilUpstream. 
Another question is still valid. My test run about 48 second. So what the time in verilator_gratt refers to/ The full time window is about 1650us?

wsnyder, 2020-10-09:
X axis is the Time Stamp Counter, https://en.wikipedia.org/wiki/Time_Stamp_Counter

If you run verilator_gantt it should print the time units.  Please attach the results of that script.

How many cycles per second is your model, or equivalently how long does one eval() call take?

simdream, 2020-10-10:
The run time Cycle Per Second is around 1.5K in single thread. After --threads 2, it turned to be 700-800. 

The above waveform seems like captured the time spend of 1.5 cycles of _eval()? from 0-1250us it is one cycle, from 1250us to 1650us is the first part of the _eval() after multi threading is done, but the reset of _eval() including "Final" are not captured. If we say the _eval() takes 1250us, this aligned well with the calculated simulation speed of 800Hz (1000ms/1.25ms = 800).

Does that mean multi threading might be more suitable for large design and complicated test case which run slower ?

====
Verilator Gantt report

Argument settings:
  +verilator+prof+threads+start+1
  +verilator+prof+threads+window+2
  --threads 2

Thread gantt graph:
  Legend: One character width = 5510 rdtsc ticks
  Legend: '&' = multiple mtasks in this period (character width)
     <-1763216 rdtsc total---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------->
  t: [xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx[xxxxxxxxxx[xxxxxx[xxxx[10-----------------] [10------][10--]                                                                                                                              [xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx[xxxxxxxxxx[xxxxx[xxxx[10--------------][10---][10---]
  t: [x[xxxxxxxxxxxxxxxxxx[xxxxxxx[xxxxx[13----]               [xxxxxxxxxxxxx[xxxxxxxxxxxx[13-------]                                                                                                                                   [xx[xxxxxxxxxxxxxxxx[13----][xxxx[13----]                [xxxxxxxxxxxx[xxxxxxxxx[13----]

Analysis:
  Total threads             = 2
  Total mtasks              = 15
  Total cpus used           = 1
  Total yields              = 0
  Total eval time           = 1763216 rdtsc ticks
  Longest mtask time        = 443124 rdtsc ticks
  All-thread mtask time     = 1853360 rdtsc ticks
  Longest-thread efficiency = 25.1%
  All-thread efficiency     = 52.6%
  All-thread speedup        = 1.1

Statistics:
  min log(p2e) = -1.41503772010834  from mtask 15 (predict 14043, elapsed 57810)
  max log(p2e) = 0.346086558979741  from mtask 5 (predict 105432, elapsed 74588)
  mean = -0.202797978417665
  stddev = 0.473358296948947
  e ^ stddev = 1.6053764814373

Writing profile_threads.vcd

wsnyder, 2020-10-10:
>Total cpus used           = 1

Both threads are running on the same physical CPU so this won't go well ;)   Use numactl as described in the docs to try to fix this.

wsnyder, 2020-10-10:
Also generally it's a good idea to give a +verilator+prof+threads+start+1 well into "normal" operation of your model as opposed to at reset time near cycle 0.

simdream, 2020-10-16:
The clock tree is complicated so that there are a lot of Vclklast update (_eval2(), _eval3(), _eval4()) after the main body of _eval() ( _eval1() in this case) is done. That kills time. Will Vclklast update be multi-threading helpful?

closed 2020-10-16 by simdream

simdream, 2020-11-13:
after changing the design slightly, the profile result is much better, just even the "all thread speedup" is 1.8, still does not see the run time speed is getting better. It just seems like the total CPU time in multi thread is getting more than single thread so no gain. Any clue how to debug further? There is no profile similar with thread=1 for a comparison.

[image]

wsnyder, 2020-11-14:
A 1.8 speedup is quite good.  It's not unusual that 2 threads won't do great compared to one, what do you get with 4, 8, 16 threads?  Be sure to use numactl.

wsnyder, 2020-11-14:
Note if you run the latest git version of verilator_gantt it'll print additional CPU info to make sure you're on the right cores.