verilator/verilator #2596: compilation problem

agrobman, 2020-10-20:
Hi,

Can you take a look, please, why GCC dies if we select large BTB structure ( a lot of flops)?
I noticed this too recently. Xrun compiles just fine ..

https://github.com/chipsalliance/Cores-SweRV-EH2/issues/11

wsnyder, 2020-10-20:
Replied to that issue, BTW if you prefer you should be able to "@wsnyder" in the SweRV repo to bring something to my attention.  (Fine this way too.)

agrobman, 2020-10-21:
@wsnyder , could you check, please, high_perf target of EH2 compilation? I had to kill it after 30min with 4.102.
Compilation was stuck on *_11.cc file, I think. It was containing the BTB logic, I think...

wsnyder, 2020-10-24:
profile=high_perf takes 4:48 cpu time, 0:54 wall time.  Verilator 4.102, GCC 9.3.0.

wsnyder, 2020-10-24:
That included verilate, compile, and run.

agrobman, 2020-10-24:
We found recently that the fpga_optimize makes compilation time long ...
add to swerv make command :
make -f $RV_ROOT/tools/Makefile **CONF_PARAMS=set=fpga_optimize=1 target=high_perf**

wsnyder, 2020-10-24:
Still fast for me. 

```
diff --git a/t/t_cores_eh2_cmark.pl b/t/t_cores_eh2_cmark.pl
index 39b354b..d83c1bf 100755
--- a/t/t_cores_eh2_cmark.pl
+++ b/t/t_cores_eh2_cmark.pl
@@ -18,9 +18,9 @@ setenv('VERILATOR', "$ENV{VERILATOR_ROOT}/bin/verilator");
 # configuration. Running from ICCM is faster and hopefully more exciting.
 # Note the build happens in $Self->{obj_dir} as the SweRV build system can
 # find everything via RV_ROOT. This leaves the submodule clean.
-run(cmd => ["make -j4 -C $Self->{obj_dir} -f $ENV{RV_ROOT}/tools/Makefile",
+run(cmd => ["make -j32 -C $Self->{obj_dir} -f $ENV{RV_ROOT}/tools/Makefile",
             "VERILATOR='$ENV{VERILATOR} ".join(' ',$Self->driver_verilator_flags()),"'",
-            "CONF_PARAMS=-iccm_enable=1",
+            "CONF_PARAMS='-iccm_enable=1 set=fpga_optimize=1 target=high_perf'",
             "GCC_PREFIX=none TEST=cmark_iccm",
             "VERILATOR_MAKE_FLAGS=VM_PARALLEL_BUILDS=1 verilator"],
     logfile => "$Self->{obj_dir}/sim.log",

```

agrobman, 2020-10-24:
@wsnyder , can you run it outside the script? - I'm not sure about ' and " in the same string in perl and what is VERILATOR_MAKE_FLAGS, we don't have? 
Also does your run script clean working directory? our make does not compile if the model was compiled already in the same directory ..

wsnyder, 2020-10-24:
I believe quoting is correct.  However as I think it looks fine at my end you should run the experiment, and should use verilator_ext_test as then we are sure it matches the nightly setup:

```
git clone https://github.com/verilator/verilator_ext_tests.git
cd verilator_ext_tests
git submodule update --recursive --init
# Instructions from .travis.yml
export VERILATOR_ROOT=`pwd`/submodules/verilator
pushd $VERILATOR_ROOT
autoconf
./configure
make -j
popd
#
t/t_cores_eh2_cmark.pl
```

agrobman, 2020-10-24:
Sorry, my fault:  I missed '-' in the set option. command should be:
make -f $RV_ROOT/tools/Makefile CONF_PARAMS=**-set**=fpga_optimize=1 target=high_perf

However, seems slow compile happens only with our new RTL, not with github version. 

I'm not sure how can we debug this problem now, before new EH2 release.

wsnyder, 2020-10-24:
If it was my decision I'd hold the release.

Anyhow to debug, perhaps run a "git bisect" then you can at least find what change caused the explosion.

agrobman, 2020-10-24:
there is a lot of changes - mostly for power savings . There is no run/compile time change with commercial tools, though.
We plan github  release next month, so we still have time. Maybe we can make privet pre-release for you ...

wsnyder, 2020-10-24:
Good to have time.  Please run a git bisect as suggested, then you'll know the exact change.

agrobman, 2020-10-24:
It gave me commit which added fpga_optimize to swerv.config :(

wsnyder, 2020-10-25:
>It gave me commit which added fpga_optimize to swerv.config :(
Suggest patching that to the makefile post-checkout (so it looks like earlier revs had it) then you should be able to find the earlier commit.

agrobman, 2020-10-28:
@wsnyder , I've found another data point - EH1
compilation-run  time was 5m37s  for our
make -f $RV_ROOT/tools/Makefile TEST=cmark_dccm

and 2m34s for
make -f $RV_ROOT/tools/Makefile TEST=cmark_dccm CONF_PARAMS=-unset=fpga_optimize

with 4.102
GCC was stuck  ( compiled too long) Vtb_top__3.cpp file  during 1st run

wsnyder, 2020-10-28:
The Travis-CI works with these, so it's something you'll need to debug on your end.  Perhaps an old gcc?  Does clang work?

agrobman, 2020-10-28:
I have no idea who Travis-CI is and know nothing about clang usage with verilator. But the latest EH2 compiles more than 30mins with this setting. I was thinking this is tool developer interest to find reason of long compilation , before users start to complain ...
I have my backup solution - I use VCS and Xelium and pretty happy with them  for all my development needs ..

wsnyder, 2020-10-28:
Travis-CI is a continuous integration tool that runs every night with Verilator against EH1 and EH2. You can see the runs here https://travis-ci.com/github/verilator/verilator_ext_tests/builds

This is only a dual CPU machine, you can see it's running every test (again including EH1 and EH2) in generally under 20 minutes.

Also as I said above, it works fine for me.

Therefore I have no way to reproduce it.

Furthermore I'm not paid by WD or Chips, so any involvement is out of the good of my heart, versus being paid like yourself to deliver a product. It you and your management don't want to support Verilator, that's your prerogative.

Therefore I'm not sure why you seem to be getting upset, I'm honestly completely unsure what you think I am able to do.

agrobman, 2020-10-28:
I'm upset waiting 30min instead of 2 out blue because of small verilog change. I never know what to expect from the verilator next. I don't use verilator for every day job, only when we are trying to release our cores to open source community. But every time I touch it I get into some trouble. Also we provide swerv cores for free and asked to use verilator as a free simulation tool. I was told you want to work with us on this open source project, but seems you have no any interest .
I also believed that solving this problem can lead to better tool - compilation time is a big problem for use verilator as RTL development tool.

wsnyder, 2020-10-29:
Sorry you feel this way.  I certainly have interest, I just have no means of helping I can determine as it works for me.  I don't know how to debug a problem I cannot see.  Again, I suggest using clang instead of gcc.

Separately I've sent an offline message to see if we can get you some assistance internal to WD.

agrobman, 2020-10-29:
I don't understand why you can't reproduce this for EH1. I gave you 2 commands which exhibit the problem .. 
"
compilation-run time was 5m37s for our
make -f $RV_ROOT/tools/Makefile TEST=cmark_dccm

and 2m34s for
make -f $RV_ROOT/tools/Makefile TEST=cmark_dccm CONF_PARAMS=-unset=fpga_optimize
"
I agree that you can't work with  EH2  on this problem, EH1 is effected too, but in smaller degree ...

wsnyder, 2020-10-29:
>with 4.102
>GCC was stuck ( compiled too long) Vtb_top__3.cpp file during 1st run

My travis comment above was related to this, it must be working OK or travis would fail much faster.

As to times:

>ccache -c
>make -f $RV_ROOT/tools/Makefile TEST=cmark_dccm
user    2m8.095s
with clang:   user  0m24.350s

>ccache -c
>make -f $RV_ROOT/tools/Makefile TEST=cmark_dccm CONF_PARAMS=-unset=fpga_optimize
user    1m30.737s
with clang:   user  0m33.653s

clang is much faster, hence you should try that.

GCC certainly is much worse, that's probably a gcc bug, but there might be something Verilator can change to work around it, I'll look into that.

agrobman, 2020-10-29:
How do I use clang with verilator?

wsnyder, 2020-10-29:
When you build verilator:

./configure CXX=clang++

then run all other steps and install.

wsnyder, 2020-10-29:
BTW if you simply try to change to clang++ in the makefile without configuring like this you'll get errors, as the compilers unfortunately don't have the exact same flags.

wsnyder, 2020-10-29:
Change this line in your makefile from -O2 to -Os. This works around GCC going insane, which I'm debugging further.

VERILATOR_MAKE_FLAGS = OPT_FAST="-Os"

With this:
fpga     gcc user    0m48.316s
nofpga gcc user    0m39.973s

agrobman, 2020-10-29:
the last one (-Os) made my day - for EH2 compile/run times are 4m13s and 4m29s for No FPGA / FPGA options,
which is much better. I also asked IT to build verilator with clang - see what do they have to say ...

BTW, where do you run ? your run times way smaller than ours ...

wsnyder, 2020-10-29:
AMD Ryzen 9 3950X.
gcc 9.3.0

It's definitely a GCC bug, will file it with them, but it'll be a long time before it reaches a distribution as fixed.  Until I have details from them I don't think Verilator can do much.

agrobman, 2020-10-29:
we use the same GCC version too. Are you running on your PC ?

wsnyder, 2020-10-29:
Yes.

wsnyder, 2020-10-29:
https://gcc.gnu.org/bugzilla/show_bug.cgi?id=97623

agrobman, 2020-10-29:
Thanks for taking this problem seriously !

wsnyder, 2020-12-06:
GNU has fixed the bug, targeting GCC 9.4 resolution.

closed 2020-12-06 by wsnyder

wsnyder, 2022-05-27:
GNU GCC team indicates fixed in 10.3.