verilator/verilator #2597: An error is reported during the compilation of bit width conversion in version 4.100

flex-liu, 2020-10-21:
verilog like this:
```verilog
module twocounter(
   input clk1,
   output wire out2
);
typedef logic [9:0] apb_offset_t;
typedef enum apb_offset_t {
  APB_OFFSET_DEBUG_DBGWVR0_EL1  = apb_offset_t'('h004 >> 2),
  APB_OFFSET_DEBUG_DBGWVR0_EL1_H1  = apb_offset_t'('h804 >> 2),
  APB_OFFSET_DEBUG_DBGWCR0_EL1  = apb_offset_t'('h808 >> 2)
} apb_offset_debug_enum;

  apb_offset_t apb_wvr_offset_hi;
  assign apb_wvr_offset_hi = apb_offset_t'(APB_OFFSET_DEBUG_DBGWVR0_EL1_H1 + apb_offset_t'('h4)  );

endmodule
```
compile log like this:
```
%Error: twocounter.v:8:3: Overlapping enumeration value: 'APB_OFFSET_DEBUG_DBGWVR0_EL1_H1'
                        : ... In instance twocounter
    8 |   APB_OFFSET_DEBUG_DBGWVR0_EL1_H1  = apb_offset_t'('h804 >> 2),
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
        twocounter.v:7:3: ... Location of original declaration
    7 |   APB_OFFSET_DEBUG_DBGWVR0_EL1  = apb_offset_t'('h004 >> 2),
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

wsnyder, 2020-10-21:
You must have WIDTH warnings disabled because it indicated it was converting these:

```
%Warning-WIDTH: t/t_EXAMPLE.v:6:58: Operator SHIFTR expects 10 bits on the LHS, but LHS's CONST '?32?h804' generates 32 or 12 bits.
    6 |   APB_OFFSET_DEBUG_DBGWVR0_EL1_H1  = apb_offset_t'('h804 >> 2),
%Warning-WIDTH: t/t_EXAMPLE.v:7:55: Operator SHIFTR expects 10 bits on the LHS, but LHS's CONST '?32?h808' generates 32 or 12 bits.
    7 |   APB_OFFSET_DEBUG_DBGWCR0_EL1  = apb_offset_t'('h808 >> 2)
```

Anyhow this should be a warning but should be fixed to result in correct values.  A workaround until fixed would be to size before the cast e.g. 

```
   localparam s1 = 'h004 >> 2;
   localparam s2 = 'h804 >> 2;
   localparam s3 = 'h808 >> 2;
typedef enum apb_offset_t {
  APB_OFFSET_DEBUG_DBGWVR0_EL1  = apb_offset_t'(s1),
  APB_OFFSET_DEBUG_DBGWVR0_EL1_H1  = apb_offset_t'(s2),
  APB_OFFSET_DEBUG_DBGWCR0_EL1  = apb_offset_t'(s3)
} apb_offset_debug_enum;

```

yanx21, 2020-10-23:
> You must have WIDTH warnings disabled because it indicated it was converting these:
>
> ```
> %Warning-WIDTH: t/t_EXAMPLE.v:6:58: Operator SHIFTR expects 10 bits on the LHS, but LHS's CONST '?32?h804' [...]

Hi wsnyder,
I came to the same problem, here is my testcase:
```sv
module t;

typedef logic [9:0] self_t;

initial begin
    $display("output: %x", self_t'('h404 >> 2));
end

endmodule
```
I built the testcase with "--cc --Wno-WIDTH", and no any warning or error.
Screen output was:
```txt
output: 001
```
However, the correct output should be '101'.
I think it's the commit "20c906261b176d1c149863dce1e4bb860eeca79b" that may bring the problem.

wsnyder, 2020-11-09:
FIxed, thanks for your reports.

Wouldn't be shocked if there's remaining cast issues.

closed 2020-11-09 by wsnyder