verilator/verilator #2598: "Segmentation fault" with coverage on

ayewinoung, 2020-10-21:
Turn on the coverage with following flags:
VerilogCompiles : "--trace" and "--coverage"
C++/SystemC Compiles : "-DVM_COVERAGE=1" and "-DVM_TRACE=1"
Design compile and run to completion and "VCD" file is written out correctly, but when "coverage.dat" is written out it causes segmentation fault.
Pretty sure, its something I'm doing it wrong or *not* doing.  In top level sc_main(..), I copied this sniplet of code from examples, at the end of simulations.
    VerilatedCov::write("logs/coverage.dat");

1. Is there any other code I need to copy ? like to initialize VerilatedCov  or something inside sc_main ?
2. Do I need to add any code inside RTL files to enable coverage ?  I'm assuming tool will add automatic basic coverage.

Any help/pointers appreciated.

wsnyder, 2020-10-21:
Can you compile with debug, make a gdb backtrace and see where it is breaking?

ayewinoung, 2020-10-21:
I fixed the seg fault by, compiling as flat (forgot about that) and reduced the complexity of design.  Thanks for the pointers.
One more question, I post process the coverage.dat with "verilator_coverage --write-info info.txt coverage.dat" and it generate somewhat readable file, as below.

Is there a way to decode these ? like whats DA means and what the numbers behind ?

Thanks.

SF:rtl_filename.v
DA:23,88788
DA:24,10
DA:27,8
DA:28,44970
DA:29,7560
DA:31,9
DA:32,16023
DA:33,2880
DA:35,4
DA:36,15489
DA:37,2713
DA:39,6
DA:40,15622

wsnyder, 2020-10-22:
See [lcov](http://ltp.sourceforge.net/coverage/lcov.php)

closed 2020-10-22 by wsnyder