verilator/verilator #2600: Compilation time

agrobman, 2020-10-21:
Hi @wsnyder,

Appears that our (EH2) configuration define (-set fpga_optimize=1) increases compilation time (GCC part) pretty significantly
from 5min to ~30min. This is counterintuitive . This define removes clock gating logic in our register slices to speed up FPGA
synthesis. This is really strange! Could you, please, investigate?

wsnyder, 2020-10-24:
This doesn't take long for me.  Please modify verilator_ext_test's t/t_cores_eh2_cmark.pl to show the issue as as perhaps I'm setting the flag incorrectly.

agrobman, 2020-10-29:
OK, we found that the reason for that slow compile was GCC -O2 bug. Replacing it to -Os brought compilation time to normal ~4-5min.
this is GCC bug reference:
https://gcc.gnu.org/bugzilla/show_bug.cgi?id=97623

the EH2 Makefile need to be patched as:
VERILATOR_MAKE_FLAGS = OPT_FAST="-Os"

wsnyder, 2020-11-03:
As above, thanks for holding in there while this was tracked down.

closed 2020-11-03 by wsnyder