verilator/verilator #2601: Support for modport expressions

rswarbrick, 2020-10-22:
It would be nice to support modport expressions like this:

```systemverilog
interface my_if;

  logic start;        // Start the operation
  logic done;         // Operation done

  modport dut (input  .start_i (start),
               output .done_o  (done));

endinterface
```
They appear in section 25.5.4 of IEEE 1800-2017 (and probably earlier versions of the standard: I haven't checked). I know that there was a "should we support this? no" question/answer in issue #1861, but I've started a new issue to track it as a feature request.

One reason that this is nice is that it allows modules to take an interface, but still keep the "foo_i is an input; foo_o is an output" naming convention.

Presumably the "trivial rename" support used above isn't particularly difficult, and I think it would already be nice as a stepping-stone. The full syntax in the spec allows an (optional) expression inside the parens. The standard is a bit vague on this point, but I guess an input can be any valid non-empty expression and an output can be assigned to nothing or to any valid lvalue in the scope of the interface (the same as a port assignment for a module instantiation). I don't know how much more difficult this general expression support would be: I'd be interested to hear.

wsnyder, 2020-10-24:
Should be straight forward if you want to try to make a pull to support this.  Note the parser has commented out lines for this to match the needed grammar.:

```
modportSimplePort<strp>:        // IEEE: modport_simple_port or modport_tf_port, depending what keyword was earlier
                id                                      { $$ = $1; }
        //UNSUP '.' idAny '(' ')'                       { }
        //UNSUP '.' idAny '(' expr ')'                  { }
        ;

```

xXIty, 2024-11-23:
How is the status on this?

wsnyder, 2024-11-23:
None as far as I know, perhaps you'd make a pull?

xXIty, 2024-11-23:
How feasible would it be for someone who is not familiar with the project?

wsnyder, 2024-11-23:
Right now it looks straight forward, but hard to know what will be hit along the way!  Worse case we'll still be closer to having it working!

Basically first step is to get the tests in good shape and passing on other simulators, then change the verilog.y to parse that and feed it into the Modport/ModportVarRef representations in the Ast tree.  Then the code that hooks up the Modports needs to support the new expressions.

Weissnix4711, 2025-12-18:
I'm happy to take this on, although I have no familiarity with this codebase. any suggestions on where to start would be appreciated.

Changing the bison is easy enough, but then somehow `void visit(AstModportVarRef* nodep)` of V3LinkDot.cpp needs to change, but what to do with this AstNodeExpr thing, I have no idea.