verilator/verilator #2602: chandle isn't compatible with null

rswarbrick, 2020-10-22:
Looking at section 6.14 of IEEE 1800-2017, `chandle`s are supposed to interact with `null` as follows:
```systemverilog
module foo;
  chandle x;
  initial begin
    x = null;
    assert(x == null);
    assert(x != null);
    assert(x);
  end
endmodule
```

Verilator seems to treat `chandle` as an alias for a 64-bit integer (at least on my 64-bit machine). It seems to treat `null` as an alias for `1'b0`, so the code above gives the following errors (tested with a v4.040 development version, `v4.040-24-ga2f48ecb` and with `Verilator 4.038 2020-07-11 rev v4.036-114-g0cd4a57ad` from Debian):
```
%Warning-WIDTH: tmp.sv:4:7: Operator ASSIGN expects 64 bits on the Assign RHS, but Assign RHS's CONST '1'h0' generates 1 bits.
                          : ... In instance foo
    4 |     x = null;
      |       ^
                ... Use "/* verilator lint_off WIDTH */" and lint_on around source to disable this message.
%Warning-WIDTH: tmp.sv:5:14: Operator EQ expects 64 bits on the RHS, but RHS's CONST '1'h0' generates 1 bits.
                           : ... In instance foo
    5 |     assert(x == null);
      |              ^~
%Warning-WIDTH: tmp.sv:6:14: Operator NEQ expects 64 bits on the RHS, but RHS's CONST '1'h0' generates 1 bits.
                           : ... In instance foo
    6 |     assert(x != null);
      |              ^~
%Warning-WIDTH: tmp.sv:7:5: Logical Operator ASSERT expects 1 bit on the Property, but Property's VARREF 'x' generates 64 bits.
                          : ... In instance foo
    7 |     assert(x);
      |     ^~~~~~
%Error: Exiting due to 4 warning(s)
```

You can make things work for verilator by replacing `null` with `0` and adding a unary OR to the last assertion, like this:
```systemverilog
module foo;
  chandle x;
  initial begin
    x = 0;
    assert(x == 0);
    assert(x != 0);
    assert(|x); // Or just use "x != 0"
  end
endmodule
```
but those changes give code that's not allowed by the spec (which doesn't allow you to compare a `chandle` with a number, assign a number to a `chandle`, or do any bitwise operations on a `chandle`).

Locally, I've worked around this with a "chandle_null" variable:
```systemverilog
`ifdef VERILATOR
  chandle chandle_null = 0;
`else
  chandle chandle_null = null;
`endif
```
but it would be nice to get rid of the workaround.

wsnyder, 2020-10-22:
Agreed.  Simple enough to fix.

Note Verilator (still) doesn't check that chandles are only used as allowed by IEEE.

closed 2020-10-22 by wsnyder

rswarbrick, 2020-10-23:
Thanks!

> Note Verilator (still) doesn't check that chandles are only used as allowed by IEEE.

Well, that's probably a good thing since otherwise we wouldn't be able to write code that worked with old and new versions of verilator using ifdefs. :-)