verilator/verilator #2608: OverWidthError: Testbench C set input 'm_axi_awid' to value that overflows what the signal's width can fit

zdai, 2020-10-26:
Hi,
I am new to Verilator and have a very basic question: how do I assign values of specific bit-width to the input port in my testbench?
For example, my top level module has an input port of width 4 bits. I use a uint8_t type variable in my testbench to assign value to this input port and get this error: 
%Error: unknown:0: Testbench C set input 'm_axi_awid' to value that overflows what the signal's width can fit

How do I specify the variable in my testbench to avoid such Error?

thanks

Zefu

wsnyder, 2020-10-26:
This error means that you have an input port of 4 bits but you are setting an upper bit (5 through 7) to a one (i.e. a value >= 16).  Verilator requires all inputs be within the bit ranges your model specified or you may get incorrect results.

closed 2020-10-26 by wsnyder

zdai, 2020-10-26:
Thank you a lot Wilson, I ANDed my input with 0xF to make sure the value falls in the valid range and the error is gone!

Regards

Zefu