verilator/verilator #2610: prof-cfunc

simdream, 2020-10-26:
What are the correct way to make profile in single thread?

This is what I did :
1. add --prof-cfunc in verilator comand
2. obj_dir/V<TOP>.mk: add "-pg" as CXXFLAGS
3. run obj_dir/V<DUT> 

We are expecting a file called gmon.dat being generated but it is not there. Maybe something missing?

wsnyder, 2020-10-26:
You also need "-pg" in LDFLAGS.

wsnyder, 2020-10-26:
Closing as pretty sure that's the problem, but feel free to ask more here if not!

closed 2020-10-26 by wsnyder

simdream, 2020-10-26:
thanks. So any chance to add -pg into CXXFLAGS and LDFLAGS in generated .mk file? Might be some reason to leave it for users manually change the Makefile?

wsnyder, 2020-10-26:
Use

     verilator ...  -CFLAGS '-pg' -LDFLAGS '-pg'