verilator/verilator #2611: 'Internal: Unknown node type reached emitter' verilating module with struct output using --protect-lib

margej, 2020-10-27:
%Error: file.v:292:12: Internal: Unknown node type reached emitter: REFDTYPE 'DBG_CPU'
  292 |     output DBG_CPU              dbg_cpu,
      |            ^~~~~~~
%Error: Exiting due to 1 error(s)

This is with v4.102. Verilates fine without --protect-lib

Given the following example:
```
typedef struct packed {

    logic           a;
    logic  [31:0]   b;
    logic  [31:0]   c;
    logic           d;
    logic           e;
    logic  [ 7:0]   f;
    logic  [63:0]   g;
    logic           h;

} DBG_CPU;
```

if --protect-lib could just convert struct I/Os to the equivilent multi-bit wire port, that would be sufficient for the instancing module to connect that port directly to a struct of the same type (this works with VCS and Xcelium)

If this is simply an accepted limitation of --protect-lib, maybe it could be listed in the documentation

simdream, 2020-10-29:
similar to #2402

wsnyder, 2020-11-05:
See #2402.

closed 2020-11-05 by wsnyder