verilator/verilator #2612: Var or Varscope here(V3Order.cpp proceeMktask)

simdream, 2020-10-28:
V3Order.cpp
in processMtask(), when annotate the mtasks that consume/produce, why it is var not varscsope? 

            // Since we happen to be iterating over every logic node,
            // take this opportunity to annotate each AstVar with the id's
            // of mtasks that consume it and produce it. We'll use this
            // information in V3EmitC when we lay out var's in memory.
            const OrderLogicVertex* logicp = movep->logicp();
            for (const V3GraphEdge* edgep = logicp->inBeginp(); edgep; edgep = edgep->inNextp()) {
                const OrderVarVertex* pre_varp
                    = dynamic_cast<const OrderVarVertex*>(edgep->fromp());
                if (!pre_varp) continue;
                ****AstVar* varp = pre_varp->varScp()->varp();  // <==  why var or the varscope?****
                // varp depends on logicp, so logicp produces varp,
                // and vice-versa below
                varp->addProducingMTaskId(mtaskId);
            }

simdream, 2020-10-28:
This is the line:
        AstVar* varp = pre_varp->varScp()->varp();  // <==  why var or the varscope?****

wsnyder, 2020-10-28:
This is for V4EmitC, which is post-descope, so there will be no VarScopes at that time.

closed 2020-10-28 by wsnyder