verilator/verilator #2614: Internal error while using dotted cross-hierarchy references

ttambe, 2020-10-28:
Hello,

I am getting the "Internal Error" below while compiling a systemverilog code that uses dotted cross-hierarchy references. The verilator manual indicates that dotted cross-hierarchy references are supported. But the error message is pointing to a line number in the code where a dotted cross-hierarchy reference is used. The command used to compile is (I added `--debg --gdbbt` to show the errors in verbose): 

`verilator +systemverilogext+.sv +verilog2001ext+.v --cc -f ../cm0_sys/tbench_M0.vc --timescale 1ns/1ps --Wno-TIMESCALEMOD --top-module CM0_SYS  --Wno-WIDTH --Wno-UNSIGNED --Wno-CMPCONST --debug --gdbbt`

The error message is: 

```
- V3Ast.cpp:1120:     Dumping obj_dir/VCM0_SYS_018_const.tree
- V3SplitVar.cpp:1259:splitVariable: 
- V3Inst.cpp:610:     dearrayAll: 
- V3Ast.cpp:1156:     Ran out of iterations finding locationStr on BASICDTYPE 0x31a1400 <e802774> {ca810ag} @dt=this@(w1)  logic kwd=logic
%Error: Internal Error: ../cm0_sys//../cm0_sys/CM0_SYS.sv:222:13: ../V3Broken.cpp:251: Broken link in node (or something without maybePointedTo): m_cellp && !m_cellp->brokeExists()
-node: IFACEREFDTYPE 0x79da280 <e1353900> {e222an} @dt=this@(w1)iface cell=AHB_S if=ahb_slave_intf mp=source -> CELL 0x1924a00 <e1753318#> {e150aq}  AHB_S -> IFACE 0x1925900 <e402522> {f16al}  ahb_slave_intf  L5 [LIB] [1ns]
  222 | .S(AHB_S[2].source),
      |             ^~~~~~
                        ... See the manual and https://verilator.org for more assistance.
- V3Ast.cpp:1120:     Dumping obj_dir/VCM0_SYS_990_final.tree
- V3Broken.cpp:300:   Broken called under broken, skipping recursion.
- V3StatsReport.cpp:229:statsReport: 
%Error: Internal Error: Aborting since under --debug
```

The error is pointing to the dotted cross-hierarchy reference in line 222 from the code as shown below

```
212 logic UART0_RXD;    
213 logic UART0_TXD;
214 logic UART1_RXD;
215 logic UART1_TXD;
216 logic [31:0] APBSUBSYS_INTERRUPT;
217 logic WATCHDOG_INTERRUPT;
218
219 APB_SYS uAPB_SYS (
220 .HCLK,
221 .HRESETn,
222 .S(AHB_S[2].source),
223
224 .UART0_RXD,
225 .UART0_TXD,
226 .UART0_TXEN(),
227
228 .UART1_RXD,
229 .UART1_TXD,
230 .UART1_TXEN(),
231
232 .UART2_RXD,
233 .UART2_TXD,
234 .UART2_TXEN(),
235
236 .TIMER0_EXTIN,
237 .TIMER1_EXTIN,
238
239 .APBSUBSYS_INTERRUPT,
240 .WATCHDOG_INTERRUPT,
241 .WDOGRESETREQ
242 );
```

Am I missing a required argument in the verilator command line? Can you please help here? I'd note that the same code compiles without error on vcs.

The verilator version is: Verilator 4.102 2020-10-15 rev v4.102-11-g818318d

Thank you!

wsnyder, 2020-10-28:
This should work but might be difficult to debug without a test case.

What is AHB_S declared as?  Try making it not be an array, e.g. AHB_S0.source and AHBS1.source.

ttambe, 2020-10-28:
Hi, AHB_S is declared as an array as below. So, in the example below, we're indexing array element 2.

```
localparam NSLAVES = 9;
ahb_slave_intf AHB_S[NSLAVES](.HCLK, .HRESETn);
```
And ahb_slave_intf is described [here](https://github.com/whatmough/CHIPKIT/blob/master/ip/ahb/ahb_intf.sv)

wsnyder, 2020-10-28:
Does making a AHB_S0 through AHB_S8 work (manually unroll it)?  Either way could you please try to make a small test case that shows the error?

ttambe, 2020-10-29:
A small [testcase.tar.gz](https://github.com/verilator/verilator/files/5455829/testcase.tar.gz) is attached. 

I have verified that using ```.S(AHB_S0.source)``` succeeds, but ```.S(AHB_S[0].source)``` fails with the internal error reported above.  
The command below is used for the testcase in both cases:
```
verilator ./test.sv +systemverilogext+.sv +verilog2001ext+.v --cc -f testinc.vc --timescale 1ns/1ps --Wno-TIMESCALEMOD --top-module CM0_SYS_TEST --Wno-WIDTH --Wno-UNSIGNED --Wno-CMPCONST --debug --gdbbt
```
Using dotted cross-hierarchy references raises the abstraction by encapsulating all the signals, which is a big useability advantage. It would really benefit Verilator to support this feature.

ttambe, 2020-10-29:
Can a resolution on this issue be expedited for an upcoming hackathon? Can you provide an ETA?

wsnyder, 2020-10-29:
The problem is an arrayed interface with modport confuses the elaboration linking, which much later results in the error.

Not immediately obvious what a straightforward fix is, but working on it now.

wsnyder, 2020-11-02:
This is proving difficult as there's several problems stacked on each other.

wsnyder, 2020-11-09:
Should be good to go now.

closed 2020-11-09 by wsnyder

ttambe, 2020-11-09:
Thanks so much Wilson! Is there a new version / tag to pull from?

wsnyder, 2020-11-09:
git master.  Probably will release a stable new version this week.