verilator/verilator #2619: Does SystemVerilog has generalized expression indexing?

udif, 2020-11-02:
I've been reading @wsnyder 's 2010 SNUG paper today ( https://www.veripool.org/papers/Preproc_Good_Evil_SNUGBos10_paper.pdf ) inspired by it to construct a convoluted verilog macro (I'll spare you the details...)

On page 7 I ran into this quote:

_This benefit is reduced with SystemVerilog2009 which finally added the {}[] syntax. So we have two choices:_
```
`define GLOBAL_VAL 22... {`GLOBAL_VAL}[1:0]   
```
I went over the SystemVerilog 2017 standard, and couldn't find any reference to a generalized indexing operator such as {}[] .
I've even tried a small example :
```
module test;
initial
  $display("%h\n", {12345}[10:0]);
endmodule
```
And it failed miserably with a fairly recent Cadence simulator (20.03-something), as well as with Verilator :-(
What am I missing? does SystemVerilog really has this feature or not?

wsnyder, 2020-11-02:
I believe {}[] was proposed but not adopted, sorry.

closed 2020-11-02 by wsnyder

udif, 2021-01-04:
One more curious question regarding the paper above.
I was using one of the macros from the paper:
```
`define BSV_DEFINED(d) `ifdef d 1 `else 0 `endif 
```
And I was asked what `BSV_` was standing for, but couldn't figure it out...

wsnyder, 2021-01-04:
BSV was the common package prefix so used on all defines, but I don't remember what it stood for and don't see any comments to that effect, sorry.

udif, 2021-01-04:
I thought it might be related to Bluespec System Verilog, but their git tree has no reference to such defines.

wsnyder, 2021-02-18:
@udif thanks for the pointer.  On reexamination this syntax should be legal, it is in 1800-2009 as this:

```
primary ::=
       | concatenation [ '[' range_expression ']' ]
       | multiple_concatenation [ '[' range_expression ']' ]

concatenation ::=
    { expression { , expression } }
```

We'll leave this closed and discuss implementation in #2721.