verilator/verilator #2621: LHS Assignment fails because not a simple variable

rayx999, 2020-11-02:
Run below test against VCS passes.

// DESCRIPTION: Verilator: Verilog Test module
//
// Use this file as a template for submitting bugs, etc.
// This module takes a single clock input, and should either
//      $write("*-* All Finished *-*\n");
//      $finish;
// on success, or $stop.
//
// The code as shown uses a function to assign localparam
//
// **If you do not wish for your code to be released to the public
// please note it here, otherwise:**
//
// This file ONLY is placed under the Creative Commons Public Domain, for
// any use, without warranty, 2020 ____YOUR_NAME_HERE____.
// SPDX-License-Identifier: CC0-1.0

module t(/*AUTOARG*/
   // Inputs
   clk
   );
   input clk;

localparam ASG0_DEPTH           = 4;
localparam ASG0_ADDR_WIDTH      = 2;

localparam ASG0_PTR_WIDTH       = (ASG0_DEPTH == 1) ? 1 : ASG0_ADDR_WIDTH + 1;
localparam ASG0_PTR_DEPTH       = 2 ** ASG0_PTR_WIDTH;
localparam ASG0_PTR_HALF_DEPTH  = 2 ** (ASG0_PTR_WIDTH - 1);

localparam logic [2*ASG0_DEPTH-1:0][ASG0_PTR_WIDTH-1:0] ASG0_GRAY_CODE = {3'h0, 3'h1, 3'h3, 3'h2, 3'h6, 3'h7, 3'h5, 3'h4};

localparam logic [ASG0_PTR_DEPTH-1:0][ASG0_ADDR_WIDTH-1:0]  ASG0_GRAY_CODE_INDEX    = asg0_gray_code_index_f(ASG0_GRAY_CODE);

    // -----------------------------
    // build gray code length 2 * ASG0_DEPTH
    // -----------------------------

    function automatic logic [ASG0_PTR_DEPTH-1:0][ASG0_ADDR_WIDTH-1:0] asg0_gray_code_index_f
        (
        logic [2*ASG0_DEPTH-1:0][ASG0_PTR_WIDTH-1:0]        x_code
        );

        logic [ASG0_PTR_DEPTH-1:0][ASG0_ADDR_WIDTH-1:0]     y_index;
        logic [ASG0_PTR_DEPTH-1:0]                          z_fill;

        // build lookup table

        y_index     = '0;

        for (int k0=0; k0<2*ASG0_DEPTH; k0++)
            {z_fill[k0], y_index[x_code[k0]]}   = k0[ASG0_PTR_WIDTH-1:0] % ASG0_DEPTH;

        // return

        return y_index;

    endfunction

    initial begin

       logic [ASG0_PTR_DEPTH-1:0][ASG0_ADDR_WIDTH-1:0] index = asg0_gray_code_index_f(ASG0_GRAY_CODE);
       if (index !== ASG0_GRAY_CODE_INDEX) $stop;
        
       $write("*-* All Finished *-*\n");
       $finish;
    end

endmodule

Run verilator test fails. Verilator version 4.100.

bash-4.2$ t/t_LHS_simple_var.pl 
======================================================================
dist/t_LHS_simple_var: ==================================================
-Skip: dist/t_LHS_simple_var: scenario 'dist' not enabled for test
dist/t_LHS_simple_var: %Skip: Skip: scenario 'dist' not enabled for test
==SUMMARY: Passed 0  Failed 0  Unsup 0  Time 0:00
======================================================================
vlt/t_LHS_simple_var: ==================================================
	perl /localdisk/Workspace/verilator.master/test_regress/../bin/verilator --prefix Vt_LHS_simple_var ../obj_vlt/t_LHS_simple_var/Vt_LHS_simple_var__main.cpp --exe --make gmake --x-assign unique -cc -Mdir obj_vlt/t_LHS_simple_var -OD --debug-check --comp-limit-members 10 --clk clk  -f input.vc +define+TEST_OBJ_DIR=obj_vlt/t_LHS_simple_var t/t_LHS_simple_var.v    > obj_vlt/t_LHS_simple_var/vlt_compile.log
%Error: t/t_LHS_simple_var.v:35:87: Expecting expression to be constant, but can't determine constant for FUNCREF 'asg0_gray_code_index_f'
                                  : ... In instance t
        t/t_LHS_simple_var.v:54:49: ... Location of non-constant ASSIGN: LHS isn't simple variable
        t/t_LHS_simple_var.v:35:87: ... Called from asg0_gray_code_index_f() with parameters:
           x_code = [0 = 3'h4, 1 = 3'h5, 2 = 3'h7, 3 = 3'h6, 4 = 3'h2, 5 = 3'h3, 6 = 3'h1, 7 = 3'h0]
   35 | localparam logic [ASG0_PTR_DEPTH-1:0][ASG0_ADDR_WIDTH-1:0]  ASG0_GRAY_CODE_INDEX    = asg0_gray_code_index_f(ASG0_GRAY_CODE);
      |                                                                                       ^~~~~~~~~~~~~~~~~~~~~~
%Error: Exiting due to 1 error(s)
%Warning: vlt/t_LHS_simple_var: Exec of perl failed: %Error: t/t_LHS_simple_var.v:35:87: Expecting expression to be constant, but can't determine constant for FUNCREF 'asg0_gray_code_index_f'

vlt/t_LHS_simple_var: %Error: Exec of perl failed: %Error: t/t_LHS_simple_var.v:35:87: Expecting expression to be constant, but can't determine constant for FUNCREF 'asg0_gray_code_index_f'
vlt/t_LHS_simple_var: FAILED: Exec of perl failed: %Error: t/t_LHS_simple_var.v:35:87: Expecting expression to be constant, but can't determine constant for FUNCREF 'asg0_gray_code_index_f'
==SUMMARY: Passed 0  Failed 1  Unsup 0  Time 0:00
==SUMMARY: Passed 0  Failed 1  Unsup 0  Time 0:00

======================================================================
	#vlt/t_LHS_simple_var: %Error: Exec of perl failed: %Error: t/t_LHS_simple_var.v:35:87: Expecting expression to be constant, but can't determine constant for FUNCREF 'asg0_gray_code_index_f'
		make && test_regress/t/t_LHS_simple_var.pl  --vlt
TESTS DONE, FAILED: Passed 0  Failed 1  Unsup 0  Time 0:00

wsnyder, 2020-11-03:
For the Location of non-constant line being errorred:

`           {z_fill[k0], y_index[x_code[k0]]}   = k0[ASG0_PTR_WIDTH-1:0] % ASG0_DEPTH;
`

Concatenate isn't supported on the left-hand-side in the elaboration evaluator.  Please change this to be two separate statements assigning to z_fill and y_index.

This will eventually be supported but is fairly complex and may be a while.

wsnyder, 2020-11-03:
P.S. If you or another person would like to contribute a fix this would of course be appreciated, and can provide how to go about it.