verilator/verilator #2622: Dangling reference at exported function with output string

yTakatsukasa, 2020-11-03:
While debugging #2508, I found that a dangling reference.

I am willing to make a PR to fix this issue.
Any feedback is appreciated. ( I am not familiar with verilated C++ code especially about naming convention and thread safe for `--threads` ).

### Example to show the symptom
```systemverilog
module t;
   function automatic void set_abc(output string s);
      s = "abc";
   endfunction
   export "DPI-C" function set_abc;
endmodule
```

### Generated C++ code
```c++
void Vt::set_abc(const char** s) {
    VL_DEBUG_IF(VL_DBG_MSGF("+    Vt::set_abc\n"); );
    // Variables
    std::string s__Vcvt;
    // Body
    static int __Vfuncnum = -1;
    if (VL_UNLIKELY(__Vfuncnum==-1)) { __Vfuncnum = Verilated::exportFuncNum("set_abc"); }
    const VerilatedScope* __Vscopep = Verilated::dpiScope();
    Vt__Vcb_set_abc_t __Vcb = (Vt__Vcb_set_abc_t)(VerilatedScope::exportFind(__Vscopep, __Vfuncnum));
    (*__Vcb)((Vt__Syms*)(__Vscopep->symsp()), s__Vcvt);
    *s = s__Vcvt.c_str();  // <=== s__Vcvt is a local variable, so *s will be a dangling pointer!
}
```
`s__Vcvt` is a local variable, so s__Vcvt.c_str() will be invalid after exiting the function.

IEEE 1800-2017 H.8.10 requires SystemVerilog side to manage the lifetime of strings.

### Fix that I am thinking
Add a member variable `__dpiExportCvt` to the top module of VL_MODULE.
```c++
    // INTERNAL VARIABLES
    // Internals; generally not touched by application code
    Vt_dpi_export_string__Syms* __VlSymsp;  // Symbol table
    std::vector<std::unique_ptr<std::string[]>> __dpiExportCvt;  // Add this member variable
```
Note that `std::unique_ptr<std::string[]> ` is used instead of `std::unique_ptr<std::string>` because I will add unpacked array support later.

Then use the `__dpiExportCvt` in each export wrapper as below.
``` c++
    auto& __dpiExportCvt = ((Vt_dpi_export_string__Syms*) __Vscopep->symsp())->TOPp->__dpiExportCvt;
    __dpiExportCvt.clear();  // delete temporary objects created in the previous call
    __dpiExportCvt.emplace_back(new std::string[1]);  // Create the temporary object
    std::string& s__Vcvt = __dpiExportCvt.back()[0];
    // do something to s__Vcvt;
   *s = s__Vcvt.c_str();  // Now  *s is valid until the next call of any exported function
```

The returned pointer will be valid until the next call of any exported functions with `output string` because 
`__dpiExportCvt` is shared among them. I think this behavior is compliant to IEEE standard as it says:

>  If it is desired to refer to the string’s valueat some point in the future, the user shall copy the string value to memory owned by the C domain.

wsnyder, 2020-11-03:
Indeed, surprised this didn't crash somebody earlier.

As to your proposal:

>Note that std::unique_ptr<std::string[]> is used instead of std::unique_ptr<std::string> because I will add unpacked array support later.

I would think this would be created based on the particular data type, so would not be an array, when you create/support unpacked arrays then a (and only a) function using arrays would be [].

I'm not seeing why you need the dynamic management, wouldn't it just work to have:

**static** std::string s__Vcvt;

Note by moving this to the symbol table or making static I think the function is no longer reentrant/thread-safe.  The DPI isn't generally reentrant so I think that's ok, but would still use:

**static** **VL_THREAD_LOCAL** std::string s__Vcvt;

If you could make a pull request would be appreciated.

yTakatsukasa, 2020-11-03:
Thanks.
I will make a PR using `static VL_THREAD_LOCAL`.

PR #2623 (2020-11-03): Fix dangling reference (Issue #2622)
Fixes #2622

Dedicated test for this issue is not added because existing test covers the case though not revealed the issue.
Maybe I was (un)lucky to hit the symptom.

https://github.com/verilator/verilator/blob/48bcf66ac7719e6ab8d553b2abfeec40fa37334f/test_regress/t/t_dpi_arg_output_type.v#L392-L396
PR #2700 (2020-12-15): Add --asan option that enables address sanitizer to the test driver
I think it is useful to compile verilated C++ with address sanitizer enabled for finding memory issues. 
I believe I was able to fix #2622 easier if I had used asan.

It would be helpful to run tests with asan. ( Perhaps CI or cron run can run with asan enabled).

It looks about 50 tests fail if asan is enabled at my first glance.
Many of failures are memory leak, but some are heap overflow such as t_dpi_open.
closed 2020-11-03 by yTakatsukasa