verilator/verilator #2625: Verilator reports errors on unreachable (by parameter value) code

alexforencich, 2020-11-04:
I have some code that looks something like this:

    if (AXIS_PCIE_DATA_WIDTH == 64) begin
        // snip
    end else begin
        // snip
        m_axis_cc_tdata_int[AXIS_PCIE_DATA_WIDTH-1:96] = shift_axi_rdata[AXIS_PCIE_DATA_WIDTH-1:96];
    end

(whole file at https://github.com/alexforencich/verilog-pcie/blob/master/rtl/pcie_us_axi_dma_rd.v)

where `AXIS_PCIE_DATA_WIDTH` is a module parameter.  When `AXIS_PCIE_DATA_WIDTH` is set to 64, verilator reports errors:

    %Error: ../../rtl/pcie_us_axi_master_rd.v:761:44: [63:96] Range extract has backward bit ordering, perhaps you wanted [96:63]
                                                    : ... In instance pcie_us_axi_master.pcie_us_axi_master_rd_inst
      761 |                         m_axis_cc_tdata_int[AXIS_PCIE_DATA_WIDTH-1:96] = shift_axi_rdata[AXIS_PCIE_DATA_WIDTH-1:96];
          |                                            ^
    %Error: ../../rtl/pcie_us_axi_master_rd.v:761:89: [63:96] Range extract has backward bit ordering, perhaps you wanted [96:63]
                                                    : ... In instance pcie_us_axi_master.pcie_us_axi_master_rd_inst
      761 |                         m_axis_cc_tdata_int[AXIS_PCIE_DATA_WIDTH-1:96] = shift_axi_rdata[AXIS_PCIE_DATA_WIDTH-1:96];
          |   

However, this code is unreachable with that parameter value, so this should not be an error.  The same module also has cases that trigger "extracting X bits from only Y bit number" from similarly unreachable lines.

wsnyder, 2020-11-04:
I agree these should probably not be shown, but the current elaboration process works on the entire design including parameter suppressed code. This is intended to be fixed eventually (#1540) but is not currently a high priority.

alexforencich, 2020-11-04:
Is this just a lint thing, or does it prevent the compilation process from continuing? I know linter warnings can be suppressed with -Wno-OPTION, is this also possible for linter errors?

wsnyder, 2020-11-04:
It's an error but you could hack to avoid it in the given statement, but something like this that makes the expression by itself legal should work:

    m_axis_cc_tdata_int[AXIS_PCIE_DATA_WIDTH==64 ? 127 : AXIS_PCIE_DATA_WIDTH-1 : 96]

alexforencich, 2020-11-04:
I see, I guess I will hold off on doing anything with verilator until this is fixed. I have quite a few modules with this sort of issue.

alexforencich, 2020-12-08:
I would love to do some more extensive experimentation with verilator, but I'm opposed to cluttering my code with ugly hacks to work around this bug.  If you don't want to fix this properly in a reasonable time frame, then please downgrade the error to a linter warning so it can be ignored.  It's incorrect as an error anyway, as the Verilog LRM does not require a specific ordering:

> Both the msb constant expression and the lsb constant expression shall be constant integer expressions. The msb and lsb constant expressions may be any integer value — positive, negative, or zero. The lsb value maybe greater than, equal to, or less than the msb value.

[Verilog 2005 LRM, section 4.3.1]

wsnyder, 2020-12-08:
> It's incorrect as an error anyway, as the Verilog LRM does not require a specific ordering

Verilator is correct this is an error (if the parameters enabled the code). If you declare as a given big or little endian it must match, IEEE 1800-2017: " The first expression shall address a more significant bit than the second expression."

>then please downgrade the error to a linter warning

But I agree with this point given the error is not always correct.

Could you provide a little test that shows the error?  I tried the attached but it doesn't error out, and your example is quite big and not self-contained.

[t_param_litendian.v.txt](https://github.com/verilator/verilator/files/5662409/t_param_litendian.v.txt)

phhorrein, 2021-01-13:
It does not error out because it's a generate (when doing this in generates, there is no issue). I have a similar issue, it happens when you use the (constant) value in the test. I created a small test.
[Uploading t_param_test.v.txt…]()

This one errors out on shiftreg, not on shiftreg_gen.

wsnyder, 2021-01-14:
If the if statement is outside the generate the code inside the if is still present, so having the out of bounds is technically illegal.   Note another commercial simulator throws a warning in this case too.

alexforencich, 2021-01-14:
Yeah, the whole point is that this error should be suppressed if the code is unreachable.  Not just in a generate block, but inside of a block that is disabled by parameter value.  If it's reachable, then a warning is definitely expected, and an error is somewhat reasonable.

phhorrein, 2021-01-14:
I think the error if it's reachable is acceptable, but only if it's reachable. In my view, this is a compile time warning (as constants have not yet been found), and an elaboration time error (as we know at that time that we are doing something wrong). 

More generally, it would be interesting to be able to downgrade errors to warnings, even if it means that something unrelated fails lated. While I agree it is not at all a best practice, for debugging/migration purpose, it could be useful.

wsnyder, 2021-01-14:
Agreed demoting to warning seems senseable - will review the pull request and thanks for working on this.

Note any named error can be demoted to a warning with -Wwarn-MESSAGENAME.  Not all errors have been named (yet) but easy to add names where needed.

PR #2754 (2021-01-15): Generate warning instead of errors for potentially unreachable code
When using a "if" statement inside an always block, part of the code may
be unreachable. This can be used to avoid errors, but it generated an
error, this commit demotes this to a warning

Partly fixes #2625
commit 3c849d7c (2021-01-15): Generate SELRANGE instead of errors for potentially unreachable code (#2625) (#2754)

When using a "if" statement inside an always block, part of the code may
be unreachable. This can be used to avoid errors, but it generated an
error, this commit demotes this to a warning. Partly fixes #2625.