verilator/verilator #2626: verilator works incorrectly with DPI-C

poemonsense, 2020-11-04:
It seems verilator works inconsistently with the following two types of DPI-C function calls.
ram.v
```
import "DPI-C" function void ram_helper
(
  input  longint    rIdx,
  output longint    rdata,
);

import "DPI-C" function longint ram_helper_1
(
  input  longint    rIdx
);

module RAMHelper(
  input             clock,
  input      [63:0] rIdx,
  output reg [63:0] rdata
);
  always @(posedge clock) begin
    ram_helper(rIdx, rdata);
  end
endmodule

module RAMHelper_1(
  input             clock,
  input      [63:0] rIdx,
  output reg [63:0] rdata
);
  always @(posedge clock) begin
    rdata <= ram_helper_1(rIdx);
  end
endmodule
```

I imported two DPI-C functions with the same behaviour:
ram.cpp:
```
#include <stdio.h>
#include <stdint.h>

#define RAMSIZE (128 * 1024 * 1024)
static uint64_t ram[RAMSIZE / sizeof(uint64_t)];

void init_ram() {
  for (int i = 0; i < 100; i++) {
    ram[i] = i;
  }
}

extern "C" void ram_helper(uint64_t rIdx, uint64_t *rdata) {
  *rdata = ram[rIdx];
}

extern "C" uint64_t ram_helper_1(uint64_t rIdx) {
  return ram[rIdx];
}
```

As far as I know, these two DPI-C function calls should return the same result. 
However, when I add two groups of FFs after them, they give different outputs. 
test.v:
```
module test(
  input         clock,
  input         reset,
  output [63:0] data,
  output [63:0] data_reg,
  output [63:0] data_1,
  output [63:0] data_reg_1,
  output [63:0] count
);

  reg [63:0] raddr;
  always @(posedge clock) begin
    if (reset)
      raddr <= 64'h0;
    else
      raddr <= raddr + 64'h1;
  end

  RAMHelper   ram   (clock, raddr, data);
  RAMHelper_1 ram_1 (clock, raddr, data_1);

  reg [63:0] data_ff, data_ff1;
  always @(posedge clock) begin
    data_ff <= data;
    data_ff1 <= data_1;
  end

  assign data_reg   = data_ff;
  assign data_reg_1 = data_ff1;
  assign count      = raddr;

endmodule
```
Following is my sim_main.cpp:
```
#include "Vtest.h"
#include "verilated.h"

void init_ram();

int main(int argc, char** argv, char** env) {
    Verilated::commandArgs(argc, argv);
    Vtest* top = new Vtest;

    uint64_t ticks = 0;
    init_ram();

    top->clock = 0;
    top->reset = 1;

    while (!Verilated::gotFinish()) {
        if (ticks == 10) {
            top->reset = 0;
        }

        top->eval();
        top->clock = ~top->clock;

        if (!top->clock) {
            printf("raddr: 0x%lx, rdata: 0x%lx, rdata_reg: 0x%lx, rdata_1: 0x%lx, rdata_1_reg: 0x%lx\n", top->count, top->data, top->data_reg, top->data_1, top->data_reg_1);
        }

        ticks++;
        if (ticks == 20) {
            break;
        }
    }

    delete top;
    exit(0);
}
```

Here is the simulation output:
```
raddr: 0x0, rdata: 0x0, rdata_reg: 0x0, rdata_1: 0x0, rdata_1_reg: 0x0
raddr: 0x0, rdata: 0x0, rdata_reg: 0x0, rdata_1: 0x0, rdata_1_reg: 0x0
raddr: 0x0, rdata: 0x0, rdata_reg: 0x0, rdata_1: 0x0, rdata_1_reg: 0x0
raddr: 0x0, rdata: 0x0, rdata_reg: 0x0, rdata_1: 0x0, rdata_1_reg: 0x0
raddr: 0x0, rdata: 0x0, rdata_reg: 0x0, rdata_1: 0x0, rdata_1_reg: 0x0
raddr: 0x1, rdata: 0x0, rdata_reg: 0x0, rdata_1: 0x0, rdata_1_reg: 0x0
raddr: 0x2, rdata: 0x1, rdata_reg: 0x1, rdata_1: 0x1, rdata_1_reg: 0x0
raddr: 0x3, rdata: 0x2, rdata_reg: 0x2, rdata_1: 0x2, rdata_1_reg: 0x1
raddr: 0x4, rdata: 0x3, rdata_reg: 0x3, rdata_1: 0x3, rdata_1_reg: 0x2
raddr: 0x5, rdata: 0x4, rdata_reg: 0x4, rdata_1: 0x4, rdata_1_reg: 0x3
```

**Now here is the problem.** 
We can see  that data and data_1 (directly produced by the two DPI-C function calls) evaluate to the same result. 
However, data_reg and data_reg_1 (directly produced from data and data_1, which should be the same) are different.

I guess, for the first form of DPI-C function `ram_helper(rIdx, rdata)`, Verilator incorrectly decides the order between data_reg and data. Maybe Verilator don't see that the DPI-C function call will change rdata's value and evaluates it to the next clock cycle's value and uses the new value to assign data_reg (in which case data_reg should use data's value in the previous clock cycle). 

To further clarify the problem, one can enable the waveform and look at how data_reg and data_reg_1 change. 
It's surprising that data_reg seems to be the same as data, instead of having 1-cycle latency. 

I just pulled the master, and this my verilator version:
```
$ ./verilator/bin/verilator --version                                                                             
Verilator 4.103 devel rev v4.102-54-g75881754
```

wsnyder, 2020-11-04:
Thanks for the good example/writeup but these are not the same.

```
    ram_helper(rIdx, rdata);
    rdata <= ram_helper_1(rIdx);
```

The former statement will assign to rdata immediately which may race or not (it's the simulator's choice), the latter has a delayed assignment.  To make them equal the task needed

```
  logic [63:0] data_t;
   always @* begin
      ram_helper(raddr, data_t);
   end
   always @(posedge clk) begin
      data <= data_t;
   end
```

Then I believe you'll get correctly identical data.

closed 2020-11-04 by wsnyder

poemonsense, 2020-11-05:
Thanks for your explanation