verilator/verilator #2628: support $size(interface_array)

nndurj, 2020-11-05:
`$size(interface_array)` doesn't work currently. Would be nice to have it.

## Example
```systemverilog
interface intf;
    logic a;
endinterface

module t (
    input v_clk
);
    intf intf_array [3:0]();
    initial begin
        $display("size:%d", $size(intf_array));
    end

endmodule
```
## Error
```
%Error: t.sv:11:35: Found definition of 'intf_array' as a CELL but expected a variable
   11 |         $display("size:%d", $size(intf_array));
      |                                   ^~~~~
%Error: Exiting due to 1 error(s)
```

wsnyder, 2020-11-05:
Perhaps you'd be willing to contribute a pull request for this?

nndurj, 2020-11-05:
@wsnyder yes, I'd like to work on it. Would like to get suggestions on how to implement this.

wsnyder, 2020-11-05:
Great, first please make a test_regress style test that checks the interface array (intf_array), non-arrayed interface and cell (intf - though not sure if this is legal). Make sure it passes on one ideally two other simulators.

Then V3Width has the code implementing $size (DIM_SIZE), probably need a new case statement item.

nndurj, 2020-11-05:
@wsnyder Thanks. I'll try that.