verilator/verilator #2629: Two dimensional array parameter is not assumed as constant

eunchan, 2020-11-05:
## Problem

Verilator assumes a parameter of two-dimensional unpacked array as a variable not a constant.

## Example

Please take a look at [Keccak Verilog HDL code here](https://github.com/lowRISC/opentitan/blob/233ea81e5a94b89edd5aa8aea3b6862080a0a3ec/hw/ip/kmac/rtl/keccak_2share.sv#L218)

The `RhoOffset` is defined as two-dimension integer type array. It is used at [L233](https://github.com/lowRISC/opentitan/blob/233ea81e5a94b89edd5aa8aea3b6862080a0a3ec/hw/ip/kmac/rtl/keccak_2share.sv#L233). Verilog lint complains (error out) that `RhoOffset[x][y]` is variable.

## Workaround

If I define the `RhoOffset [5][5]` to `RhoOffset [25]` and change the line 223 to `RhoOffset[5*x+y]%W`, Verilator does not complain the above issue.

## Verilator version

verilator 4.040 2020-08-15 rev v4.040

CC: @imphil @timothytrippel

wsnyder, 2020-11-05:
Perhaps you'd be willing to work on a pull request for a self-checking test_regress test on this, and attempted fix (probably just need V3Simulate.h improvements)?  Thanks.

dpetrisko, 2020-11-09:
Duplicate of #2272 ?

wsnyder, 2020-11-09:
Agreed, see #2272.

closed 2020-11-09 by wsnyder