verilator/verilator #2633: Optimize AND decoders

wsnyder, 2020-11-08:
AND-NOT decoders like this in C++:

```
  | (((((((((     (ib >> 0xf)
             & (~ (ib >> 0xe)))
            &     (ib >> 0xd))
         &     (~ (ib >> 0x9)))
```

from verilog e.g.

`  ib[15] & !ib[14] & ib[13] & !ib[9]
`

should be optimized into mask-comparisons e.g.

`  (ib & 'b1110001000000000) == 'b1010000000000000)
`

This is worth a few percent.

yTakatsukasa, 2021-01-05:
Let me work on this issue then #2187.

#2186 looks duplicate of this issue BTW.
I think replacing BitSel-And tree by AstEqWild as suggested in #2186 looks reasonable.

I'm wondering in which step this opt should be done.
Should it be in V3Gate ? or can it be before flattening ?
Latter sounds cheaper in computational cost.

wsnyder, 2021-01-05:
I think V3Const should do this, e.g. add a rule

       +    TREEOP ("AstAnd {matchAndTree(nodep)}",             "DONE");
 
Then matchAndTree would recurse or loop looking for a bunch of And's possibly with some Not's underneath.   Similar code (perhaps identical?) for Or, Xor.

Thanks for looking at this, made a minor test update had stuck in a branch.

yTakatsukasa, 2021-01-06:
Thank you for your advice and test update..
I found your reply while playing around V3Const. It seems I understand the code correctly.

The initial version already transforms simple tree of and-not-bitsel into eq-and.
I am going to support XOR as in 6d80e8f856044efc19006352876a43a5d03313be.

yTakatsukasa, 2021-02-21:
Resolved by #2751

closed 2021-02-21 by yTakatsukasa