verilator/verilator #2638: 'make install' chokes on src/mkinstalldirs step under cygwin

ajcrm125, 2020-11-10:
Hi, 
I was following the steps on the wiki/installing page and got all the way through until the last step.  
When running 'make install' I see the following error:

```
make[2]: Leaving directory '/cygdrive/c/Projects/verilator/src/obj_opt'
make[1]: Leaving directory '/cygdrive/c/Projects/verilator/src'
/bin/sh ./src/mkinstalldirs /usr/local/bin
./src/mkinstalldirs: line 6: $'\r': command not found
./src/mkinstalldirs: line 8: $'\r': command not found
./src/mkinstalldirs: line 10: syntax error near unexpected token `$'do\r''
'/src/mkinstalldirs: line 10: `do
make: *** [Makefile:301: installbin] Error 2

```
Seems it's complaining about carriage returns is that correct?

I'm running/installing Verilator on Cygwin:
`Verilator 4.103 devel rev v4.102-73-g8664aac2`

Anything I should try on my end?
Thanks!

wsnyder, 2020-11-11:
Almost certainly your git is misconfigured and inserting the CR's.  A search turned up e.g. 

https://stackoverflow.com/questions/26407408/dealing-with-eol-characters-in-cygwin-git-and-git-for-windows-accessing-the-same

>consider setting core.autocrlf to false

Please let us know if that or what fixed it to help the next person that hits this, thanks.

closed 2020-11-11 by wsnyder

ajcrm125, 2020-11-11:
That did the trick. I have both cygwin and MSYS on my machine so I simply switched over to MSYS and was able to complete the install. Thanks!