verilator/verilator #2640: Internal error from verilator using split_var

chrisrandall21, 2020-11-12:
bin/verilator --version
Verilator 4.103 devel rev v4.102-73-g8664aac2

bin/verilator --lint-only foo.sv
%Error: Verilator internal fault, sorry. Suggest trying --debug --gdbbt
%Error: Command Failed verilator_bin --lint-only foo.sv

foo.sv is attached.

Happy to fix with assistance.

chrisrandall21, 2020-11-12:
[foo.txt.zip](https://github.com/verilator/verilator/files/5531200/foo.txt.zip)

wsnyder, 2020-11-12:
@yTakatsukasa would you mind taking a look as backtrace says it's in V3SplitVar.  Note @chrisrandall21 also offered to help fix. Thanks.

yTakatsukasa, 2020-11-13:
I am looking into the testcase.

yTakatsukasa, 2020-11-13:
Thanks to the small enough testcase, I think I understand what's going on.

Here is the minimal testcase.
```systemverilog
module t (
  input logic var_a,
  output logic var_b,
  output logic var_c
);

  logic var_d[1:0] /*verilator split_var*/;
  assign var_d[0] = var_a;
  always_comb {var_c, var_b} = {1'h00,var_d[0]};
endmodule
```

The input AST of V3SplitVar is 018_const.tree
```
    1:2: ALWAYS 0x55ae1c85d100 <e120> {c9ad} [always_comb]
    1:2:2: ASSIGN 0x55ae1c85cd00 <e397#> {c9be} @dt=0x55ae1c859750@(G/w1)
    1:2:2:1: SEL 0x55ae1c857eb0 <e389#> {c9av} @dt=0x55ae1c859750@(G/w1)
    1:2:2:1:1: EXTEND 0x55ae1c856a50 <e359#> {c9bm} @dt=0x55ae1c859f30@(G/w2)
    1:2:2:1:1:1: ARRAYSEL 0x55ae1c85b620 <e323#> {c9bs} @dt=0x55ae1c859750@(G/w1)
    1:2:2:1:1:1:1: VARREF 0x55ae1c8563a0 <e232> {c9bn} @dt=0x55ae1c85ae50@(w1)u[1:0]  var_d [RV] <- VAR 0x55ae1c85af20 <e182> {c7aj} @dt=0x55ae1c85ae50@(w1)u[1:0]  var_d [VSTATIC]  VAR
    1:2:2:1:1:1:2: CONST 0x55ae1c856e30 <e253> {c9bt} @dt=0x55ae1c85b4f0@(G/sw1)  1'h0
    1:2:2:1:2: CONST 0x55ae1c85c3b0 <e360#> {c9av} @dt=0x55ae1c85c000@(G/w32)  32'h1
    1:2:2:1:3: CONST 0x55ae1c85ced0 <e361#> {c9av} @dt=0x55ae1c85c000@(G/w32)  32'h1
    1:2:2:2: VARREF 0x55ae1c8564c0 <e390#> {c9aq} @dt=0x55ae1c859750@(G/w1)  var_c [LV] => VAR 0x55ae1c85a0f0 <e269> {c4aq} @dt=0x55ae1c859750@(G/w1)  var_c OUTPUT [VSTATIC]  PORT
    1:2:2: ASSIGN 0x55ae1c85cbf0 <e396#> {c9be} @dt=0x55ae1c859750@(G/w1)
    1:2:2:1: ARRAYSEL 0x55ae1c85ce10 <e405#> {c9bs} @dt=0x55ae1c859750@(G/w1)
    1:2:2:1:1: VARREF 0x55ae1c857d90 <e232> {c9bn} @dt=0x55ae1c85ae50@(w1)u[1:0]  var_d [RV] <- VAR 0x55ae1c85af20 <e182> {c7aj} @dt=0x55ae1c85ae50@(w1)u[1:0]  var_d [VSTATIC]  VAR
    1:2:2:1:2: CONST 0x55ae1c85be90 <e253> {c9bt} @dt=0x55ae1c85b4f0@(G/sw1)  1'h0
    1:2:2:2: VARREF 0x55ae1c8565e0 <e394#> {c9ax} @dt=0x55ae1c859750@(G/w1)  var_b [LV] => VAR 0x55ae1c859c90 <e268> {c3aq} @dt=0x55ae1c859750@(G/w1)  var_b OUTPUT [VSTATIC]  PORT
 
```

always contains two AstAssign without BEGIN-END block.
V3SplitVar assumes AstAlways contains just single body statement. The assumption is true in Verilog level, but former pass of Verilator splits into two statements because of concat in the right hand side.

I will make a PR that PR that accepts always with multiple body statements without begin-end block.

wsnyder, 2020-11-13:
Thanks for debugging, feel free to push the patch if you think it fixes the issue & passes.

PR #2641 (2020-11-13): Fix internal error if always block without begin-end has concat in rhs (Issue #2640)
Fixes #2640 
Detail is explained in https://github.com/verilator/verilator/issues/2640#issuecomment-726779074.

I will merge tomorrow if CI passes because I'm going to bed :)
closed 2020-11-13 by wsnyder