verilator/verilator #2642: Implementing a userdefined mtask。

sequencer, 2020-11-15:
## What would you like added/supported?

I'm trying to investgate the partition algorithm based on verilator, I wanna define my own mtask like this:
```
mtask0: 
[
top/sub_instance0/sub0_instance0,
top/sub_instance0/sub0_instance1,
top/sub_instance1/sub0_instance0
]
mtask1:
[
top/sub_instance1/sub1_instance0,
top/sub_instance2/sub2_instance1,
top/sub_instance2/sub2_instance0
]
```
then use verilator to explore my algorithm.

## May we assist you in trying to fix this yourself?

I have no previous knowledge to verilator, I wanna have a basic idea in how to implementing this. If there are more assistances that would be better.

wsnyder, 2020-11-15:
Related to #2637, here's a rough outline of a way to do what you want, CC @toddstrader if he has other comments.

* Add new "/*verilator mtask <mtask_name>*/" to verilog.l
* Parse in verilog.y, setting V3FileLine mtaskHint.
* Add new mtaskAttr int in FileLine, 0 can mean not specified. This should index into a table that sets the mtaskHint based on a map-assigned number corresponding to each new name.  We don't want a string directly in FileLine as string reference counting in this critical structure would harm performance.
* Modify V3Partition (elsewhere?) to start mtask numbers it assigns above the numbers assigned above.
* Have V3Partition assign mtaskids to those requested by nodep->fileline()->mtaskAttr
* Have V3Partition not contract user-assigned mtask IDs together.

wsnyder, 2020-11-15:
@toddstrader meant to CC @jcoiner.

Related to #2637, here's a rough outline of a way to do what you want, CC @jcoiner if he has other comments.

* Add new "/*verilator mtask <mtask_name>*/" to verilog.l
* Parse in verilog.y, setting V3FileLine mtaskHint.
* Add new mtaskAttr int in FileLine, 0 can mean not specified. This should index into a table that sets the mtaskHint based on a map-assigned number corresponding to each new name.  We don't want a string directly in FileLine as string reference counting in this critical structure would harm performance.
* Modify V3Partition (elsewhere?) to start mtask numbers it assigns above the numbers assigned above.
* Have V3Partition assign mtaskids to those requested by nodep->fileline()->mtaskAttr
* Have V3Partition not contract user-assigned mtask IDs together.

wsnyder, 2021-03-12:
Closing due to age and unlikely this will get done without a champion - if you or someone starts work on this feel free to reopen.

closed 2021-03-12 by wsnyder