verilator/verilator #2643: "Strange" naming for long, escaped Verilog variables

barbara-gigerl, 2020-11-16:
Hi,

I found out that in some cases, the naming of the signals in the VCD file is not the same as in the original source code. This happens for *some* long, escaped Verilog variables although I couldn't figure out a concrete pattern:

```Verilog
//top.v
module abc(clk, rst);
  input clk;
  input rst;

  reg [1:0] \aaaaa.bbbbb.ccc.dd.eeeee[0].fffffffffff[0].ggggggg[0].hhhhh.iii.jjjj ; // produces aaaaa.bbbbb.ccc.dd____VhshdSFRsgglT92axW3QXUZOu2IIXAkusJGqilNJVt69 -> NOT OK
  reg [1:0] aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa ; // very long variable -> OK
  reg [1:0] \aaaaaaaaaaaaaaaaaaaaaaaaaaa[0]aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa ;  // very long + "[0]" in the middle -> OK
  reg [1:0] \aaaaaaaaaaaaaaaaaaaaaaaaaaa[0]aaaaaaaaaaaaaaaaaaaa.aaaaaaaaaaaaaaaa ;  // very long + "[0]" + . in the middle -> OK
  reg [1:0] \aaaaaaaaaaaaaaaaaaaaaaaaaaa[0].aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa ;  // very long + "[0]."  in the middle -> OK
  reg [1:0] \aaaaaaaaaaaaaaaaaaaaaaaaaaa[0].aaaaaaaaaaaaa[0].aaaaaaaaaaaaaaaaaaa ;  // very long + multiple "[0]."  in the middle -> OK

  always @ (posedge clk)  begin
	if(rst) begin
		\aaaaa.bbbbb.ccc.dd.eeeee[0].fffffffffff[0].ggggggg[0].hhhhh.iii.jjjj <= 0;
		aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa <= 0;
		\aaaaaaaaaaaaaaaaaaaaaaaaaaa[0]aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa <= 0;
		\aaaaaaaaaaaaaaaaaaaaaaaaaaa[0]aaaaaaaaaaaaaaaaaaaa.aaaaaaaaaaaaaaaa <= 0;
		\aaaaaaaaaaaaaaaaaaaaaaaaaaa[0].aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa <= 0;
		\aaaaaaaaaaaaaaaaaaaaaaaaaaa[0].aaaaaaaaaaaaa[0].aaaaaaaaaaaaaaaaaaa <= 0;
	end else begin
		\aaaaa.bbbbb.ccc.dd.eeeee[0].fffffffffff[0].ggggggg[0].hhhhh.iii.jjjj <= ~ \aaaaa.bbbbb.ccc.dd.eeeee[0].fffffffffff[0].ggggggg[0].hhhhh.iii.jjjj ;
		aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa <= ~  aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa ;
		\aaaaaaaaaaaaaaaaaaaaaaaaaaa[0]aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa <= ~ \aaaaaaaaaaaaaaaaaaaaaaaaaaa[0]aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa ;
		\aaaaaaaaaaaaaaaaaaaaaaaaaaa[0]aaaaaaaaaaaaaaaaaaaa.aaaaaaaaaaaaaaaa <= ~ \aaaaaaaaaaaaaaaaaaaaaaaaaaa[0]aaaaaaaaaaaaaaaaaaaa.aaaaaaaaaaaaaaaa ;
		\aaaaaaaaaaaaaaaaaaaaaaaaaaa[0].aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa <= ~ \aaaaaaaaaaaaaaaaaaaaaaaaaaa[0].aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa ;
		\aaaaaaaaaaaaaaaaaaaaaaaaaaa[0].aaaaaaaaaaaaa[0].aaaaaaaaaaaaaaaaaaa <= ~ \aaaaaaaaaaaaaaaaaaaaaaaaaaa[0].aaaaaaaaaaaaa[0].aaaaaaaaaaaaaaaaaaa ;
	end
  end

endmodule

```

The header of the produced vcd file looks like this:

```
$version Generated by VerilatedVcd $end
$date Mon Nov 16 18:02:22 2020
 $end
$timescale   1ps $end

 $scope module TOP $end
  $var wire  1 ) clk $end
  $var wire  1 * rst $end
  $scope module abc $end
   $var wire  2 # aaaaa.bbbbb.ccc.dd____VhshdSFRsgglT92axW3QXUZOu2IIXAkusJGqilNJVt69 [1:0] $end
   $var wire  2 ( aaaaaaaaaaaaaaaaaaaaaaaaaaa[0].aaaaaaaaaaaaa[0].aaaaaaaaaaaaaaaaaaa [1:0] $end
   $var wire  2 ' aaaaaaaaaaaaaaaaaaaaaaaaaaa[0].aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa [1:0] $end
   $var wire  2 & aaaaaaaaaaaaaaaaaaaaaaaaaaa[0]aaaaaaaaaaaaaaaaaaaa.aaaaaaaaaaaaaaaa [1:0] $end
   $var wire  2 % aaaaaaaaaaaaaaaaaaaaaaaaaaa[0]aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa [1:0] $end
   $var wire  2 $ aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa [1:0] $end
   $var wire  1 ) clk $end
   $var wire  1 * rst $end
  $upscope $end
 $upscope $end
$enddefinitions $end
```

I tried it with the git master version:

```bash
$ verilator -V      
Verilator 4.105 devel rev v4.104-9-g26c0e6ba
```

I would expect the name "aaaaa.bbbbb.ccc.dd.eeeee[0].fffffffffff[0].ggggggg[0].hhhhh.iii.jjjj" in the VCD file.
Also, `Vtop__stats.txt` already contains the "strange" variable name when calling Verilator like:
```bash
verilator --trace --stats-vars --trace-underscore -Wno-UNOPTFLAT -cc top.v
```
This is why I would not suspect it to be a VCD issue.

Is this a problem on my side?
If yes, can you maybe tell me how the name "aaaaa.bbbbb.ccc.dd____VhshdSFRsgglT92axW3QXUZOu2IIXAkusJGqilNJVt69" is produced? It seems not to be random, but also not follow any common encoding (like base64?).

Thank you very much for your help.

wsnyder, 2020-11-16:
Thanks for the good example, straightforward and just fixed.  Verilator hashes long identifiers to avoid breaking C++ and this "escaped".  The "dot" being a factor is only because it counts as more than one character so it pushed it over the limit - if you used a much longer name without dot same thing would occur.

BTW great avatar.

closed 2020-11-16 by wsnyder

barbara-gigerl, 2020-11-17:
Thank you for the quick response and explanation! Glad this is fixed :)