verilator/verilator #2647: Extracting structural Info for schematics

markusdd, 2020-11-17:
Short preface:
I am thinking about starting an effort for some kind of open-source schematic tracer for digital, synthesizable logic.
I feel that this is something that is substantially lacking in open source EDA-tooling right now.
Gtkwave is great, but in some cases looking at a structural schematic just makes it easier to see what is going on.
(e.g. when debugging CDC violations)

I was considering to build on the V/SV-capabilities of verilator here as it follows a very 'synthesis-like' approach and also has linting capabilities, which should make it aware of everything that a schematic representation needs.

My Question: I am not an expert of verilator internals, intermediate repesentations that might exist etc.

So, could you give me some guidance at which layer I should be looking at. I have different possibilities for tackling this in mind, each of which might have different advantages and disadvantages:

a) (Abstract/Concrete) Syntax Trees: These could generate the set of 'primitives' in the schematic. (modules, assignments/logical operators, processes etc.)
b) Connection Info: Extracting this from just Syntax is painful, if OOM references are used it is actually impossible. How does verilator represent this?
c) Structural Info: multiple drivers, clock domain of a signal etc. - Here some info must be available too, otherwise linting wouldn't work.

Grateful for any ideas on the subject.

closed 2020-11-17 by markusdd

wsnyder, 2020-11-17:
Closing as no interest after all?   BTW I was going to suggest https://github.com/alainmarcel/UHDM which aims to be a common parser for these sorts of tools and might someday front Verilator too.  Other resources are https://github.com/google/verible, https://github.com/MikePopoloski/slang and https://www.chisel-lang.org/firrtl/.

markusdd, 2020-11-17:
HI, no I mostly just closed because for some odd reason I never noticed the XML representation that can be written out by verilator, which already provides like 90% of what I might need. I must've been blind to not see this.

I have been using verible in some other context but it is still very very unstable and nothing I wish to build on right now.
UHDM is nice, but might be overkill, at least for now.

I appreciate your fast input and might reopen here if necessary. But for any basic funtionality the XML verilator writes out should be plenty.

wsnyder, 2020-11-17:
Great, was also going to point to --XML also ;)  BTW if you run with --debug you'll get .dot files that show some of the structure including connectivity - see docs/internals.adoc for more info.