verilator/verilator #2649: Failure to link nested interfaces.

dgarau, 2020-11-18:
[test_cases.zip](https://github.com/verilator/verilator/files/5557215/test_cases.zip)
Attached are the two test cases that fail to compile in verilator, both pass with ModelSim. These were derived from a more involved design that also correctly synth'd  under QuartusPro.

In the first testcase an interface that instantiates a sub interface is instantiated. A dotted reference to the sub interface is subsequently passed to a submodule port. The second testcase is similar to the first but the nested interfaces are buried under a generate for.

Hunting through the code with a debugger all the bits and pieces seem to be generated correctly. However it appears that LinkDot isn't able to resolve a link to an interface passed to a submodule unless it's in the same scope as the submodule instantiation. I attempted a patch myself but unfortunately my knowledge of the verilator dark arts was not up to the task and my 'fix' just broke a whole lot of other things.

Feel free to use the attached testcases as you wish. 
Thanks!

wsnyder, 2020-11-18:
Thanks for the good test cases and in the right format.  When I run these on two other simulators they complain that the connections are not legal, so perhaps this is only a lint checking problem, or maybe the tests got over simplfied?  Can you please check these on some other sims (e.g. "test_regress/t/t_.....pl --vcs" or --nc)

dgarau, 2020-11-18:
Thanks for the quick response!
Unfortunately Model sim is the only commercial simulator I have access to and running with --ms does indeed pass as shown below.

./t/t_interface_nested_gen_to_mod.pl -ms
==SUMMARY: Left 1  Passed 0  Failed 0  Unsup 0  Time 0:00
======================================================================
ms/t_interface_nested_gen_to_mod: ==================================================
        vlib obj_ms/t_interface_nested_gen_to_mod/work &&  vlog -sv -work obj_ms/t_interface_nested_gen_to_mod/work +define+MS=1 -ccflags "-DMS=1"  -f input.vc +define+TEST_OBJ_DIR=obj_ms/t_interface_nested_gen_to_mod t/t_interface_nested_gen_to_mod.v obj_ms/t_interface_nested_gen_to_mod/Vt_interface_nested_gen_to_mod__top.v   > obj_ms/t_interface_nested_gen_to_mod/ms_compile.log
** Warning: (vlib-34) Library already exists at "obj_ms/t_interface_nested_gen_to_mod/work".
Model Technology ModelSim - Intel FPGA Edition vlog 2020.1_3 Compiler 2020.04 Apr 27 2020
Start time: 18:59:25 on Nov 17,2020
vlog -sv -work obj_ms/t_interface_nested_gen_to_mod/work "+define+MS=1" -ccflags "-DMS=1" -f input.vc "+define+TEST_OBJ_DIR=obj_ms/t_interface_nested_gen_to_mod" t/t_interface_nested_gen_to_mod.v obj_ms/t_interface_nested_gen_to_mod/Vt_interface_nested_gen_to_mod__top.v
-- Compiling module t
-- Compiling interface interface1
-- Compiling interface interface2
-- Compiling interface interface12Array
-- Compiling module sub1
-- Compiling module sub2
-- Compiling module test
** Warning: t/t_interface_nested_gen_to_mod.v(99): (vlog-13314) Defaulting port 'inputs' kind to 'var' rather than 'wire' due to default compile option setting of -svinputport=relaxed.
-- Compiling module top

Top level modules:
        top
End time: 18:59:25 on Nov 17,2020, Elapsed time: 0:00:00
Errors: 0, Warnings: 1
        echo q | vsim -lib obj_ms/t_interface_nested_gen_to_mod/work -c -do 'run -all;quit'  top   > obj_ms/t_interface_nested_gen_to_mod/ms_sim.log
Reading pref.tcl

# 2020.1_3

# vsim -lib obj_ms/t_interface_nested_gen_to_mod/work -c -do "run -all;quit" top
# Start time: 18:59:26 on Nov 17,2020
# //  ModelSim - Intel FPGA Edition 2020.1_3 Apr 27 2020 Linux 4.19.128-microsoft-standard
# //
# //  Copyright 1991-2020 Mentor Graphics Corporation
# //  All Rights Reserved.
# //
# //  ModelSim - Intel FPGA Edition and its associated documentation contain trade
# //  secrets and commercial or financial information that are the property of
# //  Mentor Graphics Corporation and are privileged, confidential,
# //  and exempt from disclosure under the Freedom of Information Act,
# //  5 U.S.C. Section 552. Furthermore, this information
# //  is prohibited from disclosure under the Trade Secrets Act,
# //  18 U.S.C. Section 1905.
# //
# ** Warning: License feature 'alteramtivsim' will expire in 11 days.
# Loading sv_std.std
# Loading obj_ms/t_interface_nested_gen_to_mod/work.top
# Loading obj_ms/t_interface_nested_gen_to_mod/work.t
# Loading obj_ms/t_interface_nested_gen_to_mod/work.test
# Loading obj_ms/t_interface_nested_gen_to_mod/work.interface12Array
# Loading obj_ms/t_interface_nested_gen_to_mod/work.interface1
# Loading obj_ms/t_interface_nested_gen_to_mod/work.interface2
# Loading obj_ms/t_interface_nested_gen_to_mod/work.sub1
# Loading obj_ms/t_interface_nested_gen_to_mod/work.sub2
# run -all
# *-* All Finished *-*
# ** Note: $finish    : t/t_interface_nested_gen_to_mod.v(46)
#    Time: 100 ps  Iteration: 1  Instance: /top/t
# End time: 18:59:27 on Nov 17,2020, Elapsed time: 0:00:01
# Errors: 0, Warnings: 0
ms/t_interface_nested_gen_to_mod: Self PASSED
==SUMMARY: Passed 1  Failed 0  Unsup 0  Time 0:02

======================================================================
TESTS DONE, PASSED: Passed 1  Failed 0  Unsup 0  Time 0:02

wsnyder, 2020-11-18:
Perhaps you could use https://www.edaplayground.com ? 

I don't want to just post messages as we'll need a bunch of back-and-forths.

dgarau, 2020-11-18:
[tools.zip](https://github.com/verilator/verilator/files/5557602/tools.zip)
Great minds think alike!
Here are the updated test files. Now working on all the edaplayground simulators except for Cadence.

wsnyder, 2020-11-22:
Looked at this some, basically nested interfaces never worked and require some internal changes (from AstVarRef to AstVarXRef) and probably other stuff. Will give it some more time shortly.

wsnyder, 2020-11-22:
Mis-clicked close - ignore.

wsnyder, 2026-10-18:
These cases pass on master, fixed some time earlier.

closed 2026-10-18 by wsnyder