verilator/verilator #2652: Vhdl

Xyp, 2020-11-20:
Hello, 

Does anybody know a way to convert synthesizeable vhdl to verilog in a good way for usage with verilator?
I am aware of tools like vhd2vl, but those are very limited. Going over netlists is my way to address it right now, but this is rather inefficient. 

Best regards, 

Kai

wsnyder, 2020-11-20:
vhd2vl is the only option I'm aware of.

There's discussion of importing VHDL to UHDM, then feeding UHDM to Verilator but this is very premature.

There was also a thread to natively import VHDL into Verilator but this stalled.

So no great options, maybe you could contribute to one of them to add what you need?

There's also commercial translators if you can afford them.

wsnyder, 2020-11-20:
P.S. Let us know what you end up with!

closed 2020-11-20 by wsnyder