verilator/verilator #2653: Similar codes give different results.

yanx21, 2020-11-23:
Version: verilator-4.104

I have 2 tests, the only difference is the assertion of the signal `out`, but the results are different.

> test1, code:
``` verilog
module t (input   clk);

reg     in;
wire    out;
integer cyc = 0;

always @ (posedge clk) begin
    cyc <= cyc + 1;
    $write("cyc = %d  in = %x  out = %x\n", cyc, in, out);
end

always @ (posedge clk) begin
    if (cyc == 0)
        in <= 1'b1;
    else if (cyc == 1)
        in <= 1'b0;
//    else if (cyc == 2)
//        if (out != 1'b1) $stop;
//    else if (cyc == 3)
//        if (out != 1'b0) $stop;
    else if (cyc == 9)
        $finish;
end

reg internal;
assign out = internal ^ clk;
always @ (posedge clk) begin
    internal <= in;
end

endmodule
```

> test1, running result:
``` txt
cyc =           0  in = 0  out = 1
cyc =           1  in = 1  out = 1
cyc =           2  in = 0  out = 0
cyc =           3  in = 0  out = 1
cyc =           4  in = 0  out = 1
cyc =           5  in = 0  out = 1
cyc =           6  in = 0  out = 1
cyc =           7  in = 0  out = 1
cyc =           8  in = 0  out = 1
cyc =           9  in = 0  out = 1
``` 
--- 
> test2, code:
``` verilog
module t (input   clk);

reg     in;
wire    out;
integer cyc = 0;

always @ (posedge clk) begin
    cyc <= cyc + 1;
    $write("cyc = %d  in = %x  out = %x\n", cyc, in, out);
end

always @ (posedge clk) begin
    if (cyc == 0)
        in <= 1'b1;
    else if (cyc == 1)
        in <= 1'b0;
    else if (cyc == 2)
        if (out != 1'b1) $stop;
    else if (cyc == 3)
        if (out != 1'b0) $stop;
    else if (cyc == 9)
        $finish;
end

reg internal;
assign out = internal ^ clk;
always @ (posedge clk) begin
    internal <= in;
end

endmodule
```

> test2, running result:
```txt
cyc =           0  in = 0  out = 0
cyc =           1  in = 1  out = 0
cyc =           2  in = 0  out = 1
cyc =           3  in = 0  out = 0
cyc =           4  in = 0  out = 0
cyc =           5  in = 0  out = 0
cyc =           6  in = 0  out = 0
cyc =           7  in = 0  out = 0
cyc =           8  in = 0  out = 0
cyc =           9  in = 0  out = 0
```

I extracted this test from your test `t_gen_intdot`.
As you can see, test2 just uncomment the assertion code, and the output of the signal `out` are opposite.

What do you think of this results of tests?

yanx21, 2020-11-23:
Maybe I shouldn't post 2 testcase. Just test1 is enough to find out the problem.

wsnyder, 2020-11-24:
```
always @ (posedge clk)  internal <= ...
assign out = internal ^ clk;
always @ (posedge clk)  ... <= out
```

To Verilator this code races so it's up to the simulator to decide if the assign or the always evaluates first because both are sensitive to clk.  Verilator happens to make two different choices based on which it thinks is faster.  I think this is true of any IEEE simulator as both are scheduled in the active region, but would need to check on this.

In general Verilator has other restrictions related to clocks, see the manual, you can get around this by using

      wire out  /*verilator clock_enable*/;

Which tells Verilator to assume you know what you are doing and want "out" scheduled as if it's gating a clock, which will make "out" have the same edge timing as "clk".

closed 2020-11-24 by wsnyder

yanx21, 2020-12-05:
Thanks for your reply.
I tried your workaroud of adding `/*verilator clock_enable*/`, but it did not work.

What's more, I tried several simulators with the same testcases. All waveforms were exactly the same.
VCS waveform:
[image]
Mentor wavefrom:
[image]
Cadence waveform:
[image]

And they all printed the same result to the screen:
```txt
cyc =           0  in = x  out = x
cyc =           1  in = 1  out = x
cyc =           2  in = 0  out = 1
cyc =           3  in = 0  out = 0
cyc =           4  in = 0  out = 0
cyc =           5  in = 0  out = 0
cyc =           6  in = 0  out = 0
cyc =           7  in = 0  out = 0
cyc =           8  in = 0  out = 0
```

---
```
always @ (posedge clk)  internal <= ...
assign out = internal ^ clk;
always @ (posedge clk)  ... <= out
```
In my opinion, in this code, the point is not the sequent of evaluation of `assign` and `always`, but it should follow the rules:
- If `assign` evaluates first, It should use the previous value of clk, which is `0`
- If `assign` evaluates later, It should use the later value of clk, which is `1`

In other words, If `assign` evaluates first, using the later value of clk(`1`), and then `always` evaluates, indirectly using the later value of clk. but as we know, `always` must use the previous value, which contradicts the hypothesis.

Anyway, I think this issue is still valuable.

wsnyder, 2020-12-05:
"if ('x)" on all simulators but Verilator (or VCS with X-pessimism) will be false, on Verilator it's true by default and 50% of the time if randomizing.

Do you get the same result when you use initial statements to get rid of all X's?  How about changing "logic" to "bit"?

yanx21, 2020-12-06:
OK, focus on this testcase:
```verilog
module t (input   clk);

reg     in;
wire    out;
integer cyc = 0;

always @ (posedge clk) begin
    cyc <= cyc + 1;
    $write("cyc = %d  in = %x  out = %x\n", cyc, in, out);
end
always @ (posedge clk) begin
    if (cyc == 0)
        in <= 1'b1;
    else if (cyc == 1)
        in <= 1'b0;
    else if (cyc == 9)
        $finish;
end

reg internal;
assign out = internal ^ clk;
always @ (posedge clk) begin
    internal <= in;
end

endmodule
```
The output by verilator:
```
cyc =           0  in = 0  out = 1
cyc =           1  in = 1  out = 1
cyc =           2  in = 0  out = 0
cyc =           3  in = 0  out = 1
cyc =           4  in = 0  out = 1
cyc =           5  in = 0  out = 1
cyc =           6  in = 0  out = 1
cyc =           7  in = 0  out = 1
cyc =           8  in = 0  out = 1
cyc =           9  in = 0  out = 1
```
The output by other simulator (VCS, Mentor, Cadence):
```
cyc =           0  in = x  out = x
cyc =           1  in = 1  out = x
cyc =           2  in = 0  out = 1
cyc =           3  in = 0  out = 0
cyc =           4  in = 0  out = 0
cyc =           5  in = 0  out = 0
cyc =           6  in = 0  out = 0
cyc =           7  in = 0  out = 0
cyc =           8  in = 0  out = 0
```

The probrem is not the initial value, but the value after "cycle 1". Register `in` equals '1' or '0' with certainty after "cycle 1", even if the initial values differs by different simulator. So, the value of `out` should be the same after "cycle 2". However, verilator makes `out` all 1's, and other simulators make `out` all 0's.

wsnyder, 2020-12-06:
First, there are many known issues with Verilator clocking, the hope is to get these cleaned up with the new process model we are working on.  However I still don't see this is one of them.

1. You have X's in your code, which should be removed.  
2. IEEE allows a simulator to schedule a process wherever it wants even in the middle of other processes.  Verilator chooses to put the "assign out = internal ^ clk" before the write as this is better for performance.  Verilator does many things for performance that others do not.

Forcing VCS to do similar with this code:

```
module top;
logic clk; initial clk = 0;
  t t (.*);
  initial forever begin clk = ! clk; #10; end
endmodule

module t (input   clk);
  reg     in;
  reg    out;
  integer cyc = 0;

  initial in = '0;
  
  reg internal;
  initial internal = '1;

  always @ (posedge clk) begin
    cyc <= cyc + 1;
  out = internal ^ clk;
  $write("cyc = %d  in = %x  out = %x\n", cyc, in, out);
end
always @ (posedge clk) begin
    if (cyc == 0)
        in <= 1'b1;
    else if (cyc == 1)
        in <= 1'b0;
    else if (cyc == 9)
        $finish;
end
  always @* out = internal ^ clk;
always @ (posedge clk) begin
    internal <= in;
end
endmodule
```

Ggives same result as verilator (minus a time zero difference)

```
cyc =           0  in = 0  out = 0
cyc =           1  in = 1  out = 1
cyc =           2  in = 0  out = 0
cyc =           3  in = 0  out = 1
cyc =           4  in = 0  out = 1
cyc =           5  in = 0  out = 1
cyc =           6  in = 0  out = 1
cyc =           7  in = 0  out = 1
cyc =           8  in = 0  out = 1
```

Again, there are certainly many cases where verilator will get generated clocks wrong, no doubt and this is an area for improvement, but this example is likely to still show different results when that is fixed. (Although need for verilator clock_enable would be removed.)

yanx21, 2020-12-07:
OK, I retried the tests, and get the result as you said.
Thanks for your explanation that helps me a lot.