verilator/verilator #2654: Access to non overridden variable on base class

trosenkranz, 2020-11-23:
Hi Wilson,

In the attached example one cannot read (and possibly write) to a variable declared in the base class which is not overridden by the derived class. According to [this](http://www.asic-world.com/systemverilog/classes7.html) example it should be possible. A similar method call works.

I would try it but I am fairly new to ASTs and struggling with the source. Currently I suspect, that i have to extend 
[ScopeVisitor::visit(AstVar* nodep)](https://github.com/verilator/verilator/blob/103ba1fb6dd9285d16938f01c7bead5272ac6e3e/src/V3Scope.cpp#L259) or
[ScopeVisitor::visit(AstClass* nodep)](https://github.com/verilator/verilator/blob/103ba1fb6dd9285d16938f01c7bead5272ac6e3e/src/V3Scope.cpp#L150) to check whether a variable with that name exists in the class hierarchy. Any insights?

I am using verilator compiled from master from a few days ago.

Thanks Tobias

[bug_variable_access_on_base_class.zip](https://github.com/verilator/verilator/files/5585972/bug_variable_access_on_base_class.zip)

wsnyder, 2020-11-24:
Confirmed - while the class code is new thought this usage was being tested but guess not, I'll take a look.

wsnyder, 2020-11-25:
Was a big mess, should now be much better.

closed 2020-11-25 by wsnyder