verilator/verilator #2655: Cannot read 48 bit output value

KGHVerilator, 2020-11-24:
I have a module (simple simulation of a DSP48E1).

One of the ports is 
    output logic signed[47:0] p_out

My C++ code is capturing it with a uint64_t. When I check the VCD file I see all 48bits get set. However the C++ code is only able to read the least significant 32 bits.

verilator --version
Verilator 4.104 2020-11-14 rev v4.102-88-g02cb3d17a

wsnyder, 2020-11-24:
Please look at the generated .h file to make sure the port is of the type you expect e.g. a vluint64_t.

Otherwise, basic cases certainly work for this in Verilator, so please attach a self-contained example showing the problem.

KGHVerilator, 2020-11-24:
Thanks for the quick reply.
I know it generated a vluint64_t. (I cannot find where you typedefed the vluint64_t).

I will make a simple example to upload later today (my code has hacked Xilinx models so probably cannot upload them).

KGHVerilator, 2020-11-25:
Sorry, I made a mistake in sign extending from 48 bits to 64 bits.

closed 2020-11-25 by KGHVerilator