verilator/verilator #2656: Identically named interfaces across hierarchy confuses Verilator

jedbian, 2020-11-25:
Hi there,

I recently came across an issue where Verilator seems to think that identically named interfaces across hierarchy are related when they shouldn't be.

In the example below, I use a C++ testbench to drive `in_req = 0`. The expected output printed by `$display()` in `top` should always be 0, since `m1` doesn't drive `foo_if.ack` at all. However, the C++ model generated by Verilator gives the result `out_ack = 1`.

I suspect that Verilator somehow thought that `m1.bar_if` and `m2.bar_if` were the same instance.

I ran the equivalent testbench on edaplayground using QuestaSim 2020.1 and the `$display()` ouputs `'x'` as expected.

Verilator version:
```
$ verilator -V
Verilator 4.104 2020-11-14 rev v4.102-88-g02cb3d17a
```

Excample code:
```
interface a_if ();
    logic req;
    logic ack;
    modport i (
        input req,
        output ack
    );
endinterface

module m1 (
    input wire clk,
    a_if.i foo_if
);
    // Completely disconnected!
    a_if bar_if();
    always_ff @(posedge clk) bar_if.ack <= ~bar_if.req;
endmodule

module m2 (
    input wire clk,
    a_if.i bar_if
);
    m1 u_m1 ( .clk (clk), .foo_if (bar_if) );
endmodule

module top (
    input wire clk,
    input wire in_req,
    output logic out_ack
);
    a_if dut_if();

    always_ff @(posedge clk) begin
        dut_if.req <= in_req;
        out_ack <= dut_if.ack;
        $display("%d", dut_if.ack);
    end

    m2 u_dut ( .clk (clk), .bar_if (dut_if) );

endmodule
```

Output when run under QuestaSim 2020.1 on Edaplayground:
```
# Loading sv_std.std
# Loading work.top(fast)
# Loading work.a_if(fast__1)
# Loading work.m2(fast)
# Loading work.m1(fast)
# Loading work.a_if(fast__2)
# 
# vsim -voptargs=+acc=npr
# run 100 ns
#          x
#          x
#          x
#          x
#          x
```

wsnyder, 2020-11-25:
When a module gets inlined interface "bar" might misconnect to a "bar" in the upper module (that is now after inlining in the same module).

You can add a /*verilator no_inline_module*/ pragma in the appropriate place (mi) to work around this.  I'll look at fixing it - there's also a branch I have in progress to redo all of the inlining code so it might be easier to finish that off.

jedbian, 2020-11-26:
Thank you for the quick reply. I can confirm that adding the pragma workaround to either module `m1` or `m2` in my example alleviates the bug.

Looking forward to the fix!

wsnyder, 2020-12-03:
Sorry, I've put about 8 hours into this spread over the last week or so trying to fix it without the new inliner, and didn't get other tests passing with the half-fixes. I'll need to release without it and get back to it.