verilator/verilator #2657: Calling exported DPI-C function from boost coroutine

nanduraj1, 2020-11-26:
I'm not able to call an exported DPI-C function from a boost coroutine. Getting the following error
```
%Error: unknown:0: %Error: Testbench C called sv_func but no such DPI export function name exists in ANY model
```

## PFA test regress files
[dpi_coroutine.tar.gz](https://github.com/verilator/verilator/files/5604312/dpi_coroutine.tar.gz)

wsnyder, 2020-11-26:
It seems to work fine when I change it to be without boost, sorry but beyond that it's your side to debug ;)  Let us know if you get it working, and/or make a non-boost test that is failing we can debug.

BTW C++20 has a std:: coroutine library I'd encourage as an alternative, if that's an option for you.  This hasn't been debugged with Verilator either, but is likely a more stable long-term path of interest.

wsnyder, 2020-11-26:
[t_dpi_coroutine.tar.gz](https://github.com/verilator/verilator/files/5604495/t_dpi_coroutine.tar.gz)

wsnyder, 2020-12-13:
Did you figure this out - let us know as might help others.   Suspect nothing related to Verilator though so closing, reopen if otherwise please.

closed 2020-12-13 by wsnyder