verilator/verilator #2658: Travis CI is not executed recently

yTakatsukasa, 2020-11-29:
It looks that CI is not running anymore.

The last commit checked by Travis is b684995292b8943c944850b7b9eef03ecbe18912.
The next commit 103ba1fb6dd9285d16938f01c7bead5272ac6e3e was not checked.

Is this expected ? (such as during migration to other service ?)

wsnyder, 2020-11-29:
@toddstrader 

It looks like Travis has gone paid:  https://blog.travis-ci.com/2020-11-02-travis-ci-new-billing

Options:
* Regularly appeal to Travis for credits, which may mean abandoning MacOS builds as they cost more. While travis seems to say you can email regularly to get open source credits this article seems to match my feeling about this:  https://www.jeffgeerling.com/blog/2020/travis-cis-new-pricing-plan-threw-wrench-my-open-source-works
* Google Cloud is willing to contribute computes on their farm to regress Verilator. This will likely give us fastest builds but the API isn't pretty.  Example status using this: https://source.cloud.google.com/results/invocations/4e3feaa8-9688-482b-9636-dba263b4a8f6/targets 
* Travis Actions:
** https://docs.github.com/en/free-pro-team@latest/github/setting-up-and-managing-billing-and-payments-on-github/about-billing-for-github-actions
** https://developer.okta.com/blog/2020/05/18/travis-ci-to-github-actions

My inclination is towards actions, which seems the trend.  Other thoughts?

wsnyder, 2020-11-29:
Played with github actions on smaller repos, seems reasonable.

Any volunteers to convert from travis?  @gezalore @toddstrader @yTakatsukasa ?

yTakatsukasa, 2020-11-30:
I am wondering the following things:
1. FreeBSD build seems not supported on github actions
2. macOS is supported, but 10x expensive

But anyway I will try though I am not fluent in web apps.

If anyone can do it quicker, then please do so. That is great.

wsnyder, 2020-11-30:
Great!  MacOS is more expensive if a paid plan, but as open source this doesn't apply (as far as I read it anyways).

toddstrader, 2020-11-30:
Github actions seems reasonable as it is maintained by a large company and appears to have a reasonable migration path from Travis.  However, just to understand the options, is the Google Cloud situation you mentioned a free-for-FOSS thing or would this be a special setup for Verilator?  And do you have any details on what kind of compute minute / storage quotas we'd be working with?

I'm looking at the Github Actions pricing and it appears that we'd get 500 MB and 2000 minutes per month for free.  I think this would be for everything under the Verilator Github organization (so the main repo but also verilator_ext_tests).  I'll see if I can scrape any comparable data from Travis in terms of what we've been using in the past, but I'm thinking we might need to scale back testing to only run once a day (or less frequent) with these limits with Actions.

Happy to help, but I'm not sure this is something that lends itself to dividing and conquering (besides verilator_ext_tests, but we should definitely handle the main repo first and then take care of that one).  @yTakatsukasa let me know if you would like me to do anything here.  Otherwise I'll leave you to it since you got here first.

Also, @wsnyder where does it say that the multiplier doesn't affect the free plan?  I'm looking at this:

> Jobs that run on Windows and macOS runners that GitHub hosts consume minutes at 2 and 10 times the rate that jobs on Linux runners consume. For example, using 1,000 Windows minutes would consume 2,000 of the minutes included in your account. Using 1,000 macOS minutes, would consume 10,000 minutes included in your account.

which I assume is what @yTakatsukasa saw too.  Maybe I'm missing something but this sounds like it's for everyone.

toddstrader, 2020-11-30:
Yeah, our last build took a total of ~ 12 hrs (with caching) and has ~ 1.3 GB of cached data:
https://travis-ci.com/github/verilator/verilator
https://travis-ci.com/github/verilator/verilator/caches

Unless I'm reading the Actions pricing wrong we'll have to drastically reconsider what and how much CI we need.

wsnyder, 2020-11-30:
I do think I misread what options apply with GitHub.   

Not sure if we'd want to call Google services directly, or still have a single github action that then calls into the Google cloud.  I think if github called into Google we'd still be limited to 2000 minutes of regressions, but on Google cloud a complete regression will be ~5 minutes, so that's 400 builds/month  - perhaps still worth avoiding a limit on if we can.

Meantime I'll talk to Google about cloud accounting. The Google offer was for Verilator specifically, as Google is both a user and contributor.

toddstrader, 2020-11-30:
OK, cool.  Then I definitely think it's worth having that conversation first before embarking on anything else given the potentially huge difference in CI bandwidth it represents.  Also, Github has native Google Cloud integration (just like they do for Travis):
https://github.com/marketplace/google-cloud-build

So I don't think we'd need to use Github Actions to trigger builds with Google Cloud.  I could be wrong but I don't think we'd run into the 2000 minute limit if we used that.

yTakatsukasa, 2020-11-30:
OK. I pause my work. (I just browsed documents).

mithro, 2020-11-30:
From https://docs.github.com/en/free-pro-team@latest/actions/reference/usage-limits-billing-and-administration it says;
> GitHub Actions usage is free for public repositories and self-hosted runners.

> **Usage limits**
> *There are some limits on GitHub Actions usage when using self-hosted runners. These limits are subject to change.*
>  * Workflow run time - Each workflow run is limited to 72 hours. If a workflow run reaches this limit, the workflow run is cancelled.
>  * Job queue time - Each job for self-hosted runners can be queued for a maximum of 24 hours. If a self-hosted runner does not start executing the job within this limit, the job is terminated and fails to complete.
>  * API requests - You can execute up to 1000 API requests in an hour across all actions within a repository. If exceeded, additional API calls will fail, which might cause jobs to fail.
>  * Job matrix - A job matrix can generate a maximum of 256 jobs per workflow run. This limit also applies to self-hosted runners.

I can also set up Google funded "self hosted runners" on GCP for Verilator too (which we probably want to do to speed up the regression suite run time anyway).

mithro, 2020-11-30:
But I have never used self hosted runners and public repositories have certain security implications I don't understand now.

wsnyder, 2020-11-30:
@mithro I believe what you quoted are completely reasonable GitHub limits well under how much verilator would use actions provided a self-hosted runner.  It would be awesome and much appreciated if Google could provide the runner system resources, this would solve the compute problem, would let us run more than 2 jobs to better test multithreaded, and would speed up the turn-around time from push commit to job completion.  We can figure out the provisioning and actions and such if you'd prefer not to. Could you ask about the cloud funding - email me if there's some credentials/justification/account I need to setup/etc that would help out!

mithro, 2020-12-01:
I think I mangled that quote, it should be;

> ## Usage limits
> There are some limits on GitHub Actions usage when using **GitHub-hosted** runners. These limits are subject to change.
>
> *Note: **For self-hosted runners, different usage limits apply**. For more information, see "About self-hosted runners."*
>
>  * Workflow run time - Each workflow run is limited to 72 hours. If a workflow run reaches this limit, the workflow run is cancelled.
>  * Job queue time - Each job for self-hosted runners can be queued for a maximum of 24 hours. If a self-hosted runner does not start executing the job within this limit, the job is terminated and fails to complete.
>  * API requests - You can execute up to 1000 API requests in an hour across all actions within a repository. If exceeded, additional API calls will fail, which might cause jobs to fail.
>  * Job matrix - A job matrix can generate a maximum of 256 jobs per workflow run. This limit also applies to self-hosted runners.

aka -- GitHub is providing the listed resources above to any **public** repo for free. The minutes restriction is only on **private** repos.

We still might want to have self hosted runners on Google Cloud in the future but I'm still investigating how you do that (issue is not providing the resources but how the whole thing actually works).

mithro, 2020-12-01:
So, what I'm saying is proceed to migrate to GitHub Actions (it'll give you more free resources than Travis CI use to) on the GitHub-hosted runners and we will follow up with the Google Cloud stuff afterwards.

umarcor, 2020-12-10:
For clarification (from experience), there are currently no limits de facto for public (say free, say open source) repositories and/or organisations in GitHub Actions. I have seen this discussion multiple times in several projects during this last year, and very specially this last month. You might have some jobs delayed (queued) or auto-cancelled if you try to run say 30 workflows (not jobs) at the same time. But that's a pretty high limit. Regarding artifacts, there is no notable limit neither.

NOTE: almost a year ago, some workflows would not start if you run over the limit. That was very misleading for users. Now, those are shown as queued.

The heaviest organisation I am aware about is Godot. It has a thriving activity and each PR/push generates several MBs of artifacts: https://github.com/godotengine/godot/actions. Still, all good so far. See the discussion about enabling artifact uploads, and the concerns about limits: https://github.com/godotengine/godot/pull/42422.

I would say that the only "problem" for migrating Travis to GitHub Actions is that some very specific features are cannot be mapped exactly. For instance "allow failure" or "create release and upload assets". Other than that, I think it's pointless for GitHub projects not to use GitHub Actions in 2020-2021, since it's the cheapest and fastest service with the best integration and it's supported by one of the largest cloud service providers. As it happened with Travis, this might change in the future, but that's not the case at the moment.

Naturally, using self-hosted runners or distributing some workloads to Google Cloud does still make sense for the workflows which cannot fit in GitHub Actions due to disk size constraints. Windows runners are unlikely to run out of space, but Linux jobs might do. That's an issue which needs to be addressed in SymbiFlow. The know-how can then be reused in this repo, or testing master Verilator can be included in whichever solution is adopted in SymbiFlow. As @mithro said, there are some constraints with enabling self-hosted runners on public repos. Therefore, a sensible solution might be to have a private repo which mirrors all the tools/projects that are to be tested intensively.

There is a very specific reason to not ditch Travis completely: ARM builds. It is possible to test ARM containers on GitHub Actions using QEMU (see [dbhi/qus](https://github.com/dbhi/qus/)). However, that's known to be 5-10 times slower than native execution. I think that using dbhi/qus for building Verilator is acceptable. Not sure about running tests, tho.

BTW, I reached this issue because I wanted to suggest the contribution guidelines to be updated in this regard:

> Verilator uses Travis CI to provide continuous integration. You may want to setup Travis CI on your GitHub branch to ensure your changes keep the tests passing. See docs/internals.

Currently, any user that forks this repo will get CI by clicking the button in their Actions tab. No need to configure anything explicitly.

wsnyder, 2020-12-10:
Thanks for your comments and "love" you are applying to clean up our early attempts at using Actions.  I think we're fully on board using actions now.  I committed fixing the internals documentation you mentioned.

wsnyder, 2020-12-10:
We're settled on actions and have great pulls in flight, so think this meta discussion has resolved, thanks all for feedback & improvements.

closed 2020-12-10 by wsnyder