verilator/verilator #2659: 1ns/500ps timescale is rejected on commandline

markusdd, 2020-11-30:
While we were investigating and solving https://github.com/cocotb/cocotb/issues/2096 together, I extended my AFIFO TB to now run 500 different, randomized clock ratios.

These are given in scale of nanoseconds, but per default everything runs on 1ps/1ps. So basically the main verilator loop runs for 1000 times per fastest clock cycle, including all callbacks, which is ridiculously slow. And unnecessary.

So I wanted to use 1ns/1ns. But this forbids clock periods like 3ns, as the falling edge ends up at 1.5ns.
So I get errors from cocotb (rightfully so) that timescale forbids properly modelling the clock which was requested.
But 1ns/500ps would be perfectly fine for this (and valid from verilog perspective).

I tried giving this to verilator on commandline, but it yells at me with a syntax error. 1ps/1ps and 1ns/1ns works perfectly.

a) Is this known/intended?
b) Is this a bug?

For a trial I restricted myself to even ns-periods for the clocks and it works like a charm when using 1ns/1ns and is blazing fast, the 500 iterations over the ratios run much faster like in icarus or even xcelium. But with 1ps/1ps it is just super, super slow.

wsnyder, 2020-11-30:
c. That isn't legal.

IEEE: "The integers in these arguments specify an order of magnitude for the size of the value; the valid integers are
1, 10, and 100. The character strings represent units of measurement; the valid character strings are s, ms,
us, ns, ps, and fs.

closed 2020-11-30 by wsnyder

markusdd, 2020-12-01:
You would be absolutely correct. I did not know that.

Yet, let me ask one thing: Would it anyway make sense to support something like this? As verilator is cycle-based, this could actually yield rather cheap performance improvements.

I do not know the exact effect on the internals though, so consider this a 'dumb' question.

wsnyder, 2020-12-01:
Verilator uses power-of-ten representations which makes things much faster, so I don't want to support non-standard values.  I also think there might be potential floating point rounding problems.  So, sorry don't want to support this at this time.

markusdd, 2020-12-01:
Sure, It was just an idea in terms of reducing the number of cycles you have to 'walk through the main loop'. 
But if this breaks other internals it might not be worth it.