verilator/verilator #2660: Concurrent/Asynchronous simulation of verilated designs

andreaskuster, 2020-12-01:
First, thanks a lot for this great and very powerful Verilog/SV simulator!

We are currently integrating Verilator into DaCe as part of the RTL CodeGen backend (https://github.com/spcl/dace). For that, we would like to simulate multiple tasklets (Verilog/Verilator design) concurrently, in separate threads. This works well until we add support for tracing, see https://github.com/spcl/dace/pull/490. 

The main problem is, that Verilator expects a single global `double sc_time_stamp()` function, which is not scalable to multiple concurrent simulations. 

Is there a way to overcome this limitation with the current implementation of Verilator?  

Otherwise, possible solutions would be:
- Add a parameter to the `double sc_time_stamp()` function indicating where the call comes from (i.e. in my case top0 or top1). 
- Make the function dependent on the design name i.e. `double sc_time_stamp_top0()`
- Add a function pointer parameter to the initialization routine, such that the user can write her/his own routine and passes it.

I attached a minimal working example that should work out-of-the box i.e `cmake . && make && ./simulation`

[multi_tasklet_issue_example.zip](https://github.com/verilator/verilator/files/5622887/multi_tasklet_issue_example.zip)

wsnyder, 2020-12-01:
That's certainly a reasonable use case.  

Would the cleanest interface be to have a switch where the timestamp is stored as a member variable in the top verilated class object (e.g. class Vtop)?

I don't know but suspect there may be additional globals that will cause problems.  Certainly the proposed timing wheel additions will break.  Let me do some auditing.

BTW double sc_time_stamp() is the old interface, better to define VL_TIMESTAMP64 and provide vl_time_stamp64().  If we add a member variable it would be uint64_t.

andreaskuster, 2020-12-01:
Yes, I agree that integrating this time/cycle count variable is semantically tied to a design and would fit well in the top verilated class (and this would certainly solve this problem).

You are right. I haven't really looked into the internals of `$finish` / `Verilated::gotFinish()`, but this seems to be handled globally too (once for all verilated designs). I was able to circumvent this issue by not observing `$finish`, but rather look at the ready/valid or out signals instead (this is also the case in the attached `multi_tasklet_issue_example.zip`).

Sure, please let me know about your findings, and I am happy to provide a PR (given your input about the most appropriate solution).

Thanks a lot for the hint about VL_TIMESTAMP64. Since I mostly relied on the examples, I suggest to update this there too (i.e. at least this https://github.com/verilator/verilator/blob/master/examples/make_tracing_c/sim_main.cpp#L17 still uses ` double sc_time_stamp()`

wsnyder, 2020-12-02:
## General concept ##

Most of what is static in Verilator would change to be stored under a simulation context.

A given Verilated model would point (perhaps on construction) to a given simulation context.  Multiple models would point to the same context when they are directly connected (e.g. a big SystemC model with many Verilated modules).

What is currently a static function like e.g. Verilated::gotFinish would change to e.g.

    bool Verilated::gotFinish() { return VerilatedContext::defaultContextp()->gotFinish(); }

and this function would only be for backward-compatible use.

All internals would call the model's e.g. __VmContextp as needed.

The downside of this approach is an indirection to get to some hot variables, most notably time().  This might not be too expensive, and is probably worth a try as is.  Alternatively the default could be to have __VmContextp be a static which would give the current global single context and no pointer dereference (as would be link-time resolved). Then a new verilation-time switch would make __VmContextp become non-static, enabling multiple contexts to exist.

We'll need some flag to enable this new API as I see no way to keep sc_time_stamp()/vl_time_stamp64() in a backward-compatible way.  (Maybe an API version selector?)

## Static stuctures ##

The following are the static structures I found:

* $finish/$stop (s_flushCbs, s_exitCbs, s_gotFinish, s_errorCount, s_errorLimit, Verilated::gotFinish(), Verilated::flushCall(), vl_finish, vl_stop, vl_stop): Change to context based.

* $time (vl_time_stamp64(), sc_time_stamp(), future event time wheel): Change to context based.

* $timeformat (m_timeFormat*, s_timeunit, s_timePrecision): Change to context based.

* Command line arguments (s_args, s_profThread*, m_argVec*, s_debug*, Verilated::commandArgs() Verilated::commandArgsPlusMatch(), Verilated::debug()): Could see as global or context based.  Probably better to make
as much context based as possible.

* File handles (m_fd*): Handling of $fopen.  Change to context based.

* Randomize (s_rand*, Verilated::randReset()): Random seed/epoch/etc. Change to context based.

* Assertion enable (s_assertOn): Change to context based.

* Tracing general (verilated_trace, verilated_vcd): API should work as-is.  Change to get time via context.

* Tracing enable (s_calcUnusedSignals, Verilated::traceEverOn()): Change to context based.

* Coverage (VerilatedCovImp, VerilatedCov::clear(), VerilatedCov::clearNonMatch(), VerilatedCov::write(), VerilatedCov::zero()): Coverage is tracked globally, called globally with e.g. VerilatedCov::write).  Eventual SystemVerilog coverage support will turn on/off coverage dynamically and time-based, so perhaps best made context-based.

* Multithreaded statistics (s_yields): Performance counter.  Matters globally and not simulation-result relevant so can
remains global.

* Save/restore (verilated_save): Called as method of the model, so no API change.  Most of what is globally saved/restored would instead do the context save/restore.  This means when multiple contexts are used each would need to be separately saved (or restored), that seems conceptually correct.  The serialization would also allow a process to
"clone" a context and model into a new context and model, which sounds potentially useful (e.g. fault simulations).

* VPI (VerilatedVpiImp, VerilatedScopeNameMap, VerilatedHierarchyMap, ExportNameMap, s_fatalOnVpiError,
Verilated::fatalOnVpiError()): VPI state is global.  The IEEE VPI design probably won't ever work with multiple contexts.  Leave global?  Or make context based so nothing static is left?

* DPI calls (V3EmitC.cpp:181,1918, m_userMapMutex, Verilated::dpiScope()): Information passed to/from DPI calls are
static.  Multiple models can't have simultanious calls into the DPI.  These statics could be moved into member variables for a small performance cost.  Context DPI calls require same global structures as VPI. Not sure how critical DPI is, as DPI isn't generally reentrant/multi-design compatible.  Leave global?  Or make context based so nothing static is left?

## Implementation ##

Possible implementation route:

1. Discuss
2. Create test_regress style test showing your multi-model use case.  (Which won't work until below)
3. Create __VmContextp, new Verilator option, move sc_time_stamp() to the new model, see how it works for your use case.
4. Move $finish/$stop, anything else highly critical for you use.
5. Move other constructs above.

Thoughts/Discussion?!

andreaskuster, 2020-12-05:
Thanks for the summary. 

Your implementation route looks reasonable to me. As you already mentioned, moving sc_time_stamp() and $finish to the class context is most critical for my usecase.

I just wrote a test_regress test for the multi model case (fails right now due to the $finish, the sc_time_stamp() is harder to model without compiler errors. I will adjust the test as soon as we have the new function call in place. 

- Do you prefer if I fork+PR or do you want to create a new branch such that we can work there?

- And how do you want to split work?

[verilator_test_regress_t_multi_model.zip](https://github.com/verilator/verilator/files/5647188/verilator_test_regress_t_multi_model.zip)

wsnyder, 2020-12-05:
You can make a fork, if you file a pull just have "allow edits from maintainers" selected.  If it gets complicated we can move it to a branch.

The more you're willing/able to do the better - but feel free to ask for help and/or when you run out of time, I or someone can take over.

wsnyder, 2021-02-28:
Please see new pull #2813 this means models can now be created under a VerilatedContext, and time is now under that structure.  The old API should all still work (hopefully).

```
    const std::unique_ptr<VerilatedContext> contextp{new VerilatedContext};
    //                                                 ^^^^^^^^^^^^^^
    contextp->commandArgs(argc, argv);
    //^^^^^^
    const std::unique_ptr<Vt_flag_main> topp{new Vt_flag_main{contextp.get()}};
    //                                                         ^^^^^^
    while (!contextp->gotFinish()) {
        topp->eval();
        contextp->timeInc(1);
       //  ^^^^^^  ^^^^^^^^
    }

```

wsnyder, 2021-02-28:
Notes on how simulation time is now retrieved by Verilator (from comments in verilated.h):

How Verilator runtime gets the current simulation time:

* If using SystemC, time comes from the SystemC kernel-defined sc_time_stamp64(). User's wrapper must not call SimulationContext::time(value) nor timeInc(value).
* Else, if SimulationContext::time(value) or SimulationContext::timeInc(value) is ever called with non-zero,
then time will come via the context.  This allows multiple contexts to exist and have different simulation times. This must not be used with SystemC.  Note Verilated::time(value) and Verilated::timeInc(value) call into SimulationContext::time and timeInc, operating on the thread's context.
* Else, if VL_TIME_STAMP64 is defined, time comes from the legacy 'vluint64_t vl_time_stamp64()' which must a function be defined by the user's wrapper.
* Else, time comes from the legacy 'double sc_time_stamp()' which must be a function defined by the user's wrapper.

andreaskuster, 2021-03-01:
Great, thanks a lot! I added my comments on #2813

closed 2021-03-07 by wsnyder