verilator/verilator #2661: Writes to arrays (potentially) getting optimized away

rachitnigam, 2020-12-01:
### Steps to reproduce

**Version**: Verilator 4.104 2020-11-14 rev v4.102-88-g02cb3d17a

Files for reproduction: [repro.tar.gz](https://github.com/verilator/verilator/files/5625211/repro.tar.gz)

1. Untar the file:
```
tar xvf repro.tar.gz
```
2. Run the build script (just runs verilator and executes the binary)
```
./gen-vcd out.sv
```
3. This generates a vcd file (`out.vcd`) and writes the following to STDOUT:
```
Simulated 4 cycles
[mem]:          0
```
4. Check the value of `main.Res0_0.mem[0][0]` in your favorite waveform viewer. It will be `0`. 

**Expected output**: Running the simulation writes
```
Simulated 4 cycles
[mem]:          20
```
and the value in `main.Res0_0.mem[0][0]` contains the value `20`.

### Investigation

The program implemented in `out.sv` exponentiates a value and writes it to a memory. The memory "primitive" is implemented as:
```verilog
module std_mem_d2
  #(parameter width = 32,
    parameter d0_size = 16,
    parameter d1_size = 16,
    parameter d0_idx_size = 4,
    parameter d1_idx_size = 4)
   (input logic [d0_idx_size-1:0] addr0,
    input logic [d1_idx_size-1:0] addr1,
    input logic [width-1:0]   write_data,
    input logic               write_en,
    input logic               clk,
    output logic [width-1:0]  read_data,
    output logic done);

  /* verilator lint_off WIDTH */
  logic [width-1:0]  mem[d0_size-1:0][d1_size-1:0];

  assign read_data = mem[addr0][addr1];
  always_ff @(posedge clk) begin
    if (write_en) begin
      mem[addr0][addr1] <= write_data;
      done <= 1'd1;
    end else
      done <= 1'd0;
  end
endmodule
```

During normal execution of the file, the `write_data` and `write_en` signals are held high for a cycle:
[image]
However, the memory (in the image `mem(0)(0)`) never shows the values.

What's weird is that the following changes all "fix" the problems:
1. Add `/* verilator public */` to `write_data`.
2. Add an unused register to `std_mem_d2` and write the `write_data` signal into it.
3. Add `/* verilator public */` to `mem`. This only makes the value appear on STDOUT and no the VCD.

I don't really understand why any of these work so I want to figure out what is going on here. My guess is that somewhere, the `write_data` signal get optimized away.

### Context
This code is generated by a HLS-like [compiler](https://github.com/cucapra/futil/). If there is some issue with us misusing `always_ff`, `logic` etc., please let us know.

wsnyder, 2020-12-01:
Verilator is a 2 state simulator so it randomizes values instead of doing X's (which is good in this case - stay tuned...)

Anyhow if write_en randomizes to 1, on the first clock edge done asserts early and the "correct" write never happens.

So basically the randomization is finding a time-zero reset bug.

Your confusion is because based on optimizations Verilator might optimize time zero behavior differently.

Basically you should call eval once before any clocks, or better assert "go" later than time zero, then it will be stable, I get the same results then tracing or not.

closed 2020-12-01 by wsnyder

rachitnigam, 2020-12-02:
Thanks for the explanation! However, it seems like adding the `/*verilator public*/` comment or adding the fake register is still changing the behavior of the design.

I changed the `testbench.cpp` file to say:
```c++
  top->eval();
  top->clk = 0;
  int done = 0;
  int ignore_cycles = 5;
  printf("Starting simulation\n");
  while (done == 0 && i < n_cycles) {
    done = top->done;
    if (ignore_cycles == 0) {
      top->go = 1;
    } else {
      ignore_cycles--;
    }
    // dump variables into VCD file and toggle clock
    for (clk = 0; clk < 2; clk++) {
      if (trace) {
        tfp->dump(2 * i + clk);
      }
      top->clk = !top->clk;
      top->eval();
    }

    if (Verilated::gotFinish())
      exit(0);

    i++;
  }
```
(I spend a few cycles doing nothing at the start.)

If I change the memory primitive to say:
```verilog
  /* verilator lint_off WIDTH */
  logic [width-1:0]  mem[d0_size-1:0][d1_size-1:0];
  logic [width-1:0]  fake;

  assign read_data = mem[addr0][addr1];
  always_ff @(posedge clk) begin
    if (write_en) begin
      mem[addr0][addr1] <= write_data;
      // THIS WRITE IS NEVER OBSERVED
      fake <= write_data;
      done <= 1'd1;
    end else
      done <= 1'd0;
  end
```
I suddenly the get the right value.

I should have mentioned this in the initial report but I am always running the design with tracing so I wasn't expecting it to make any difference (thanks for catching the issue either way!)

rachitnigam, 2020-12-02:
Some more test case reduction. This also happens if the memory is replaced with a simple register. Uncommenting line 30 gives the right answer (10) whereas the default file just gives 0:

```systemverilog
module std_exp
  (input  logic        go,
   input  logic        clk,
   output logic [31:0] out,
   output logic        done);
   always_ff @(posedge clk) begin
     if (go) begin
       out = 10;
       done = 1;
     end else begin
       done = 0;
       out = 0;
     end
   end
endmodule
module std_mem_d2
  #(parameter width = 32)
   (input logic [width-1:0]   write_data,
    input logic               write_en,
    input logic               clk,
    output logic done);

  logic [width-1:0]  mem;
  logic [width-1:0]  fake;

  always_ff @(posedge clk) begin
    if (write_en) begin
      mem <= write_data;
      // UNCOMMENT TO GET THE REAL ANSWER
      // fake <= write_data;
      done <= 1'd1;
    end else
      done <= 1'd0;
  end
endmodule
module main (
    input logic go,
    input logic clk,
    output logic done
);
    bit [31:0] Res0_0_write_data;
    bit Res0_0_write_en;
    bit Res0_0_done;

    bit exp0_go;
    bit [31:0] exp0_out;
    bit exp0_done;

    std_mem_d2 # (
        .width(32)
    ) Res0_0 (
        .clk(clk),
        .done(Res0_0_done),
        .write_data(Res0_0_write_data),
        .write_en(Res0_0_write_en)
    );

    std_exp exp0 (
        .clk(clk),
        .done(exp0_done),
        .go(exp0_go),
        .out(exp0_out)
    );

    always_comb begin
        if(go) begin
          Res0_0_write_en = exp0_done;
        end else begin
          Res0_0_write_en = 1'd0;
        end

        if (exp0_done & go) begin
            Res0_0_write_data = exp0_out;
        end else begin
          Res0_0_write_data = 32'd0;
        end

        if (Res0_0_done) begin
            done = 1'd1;
        end else begin
          done = 1'd0;
        end

        if(~exp0_done & go) begin
            exp0_go = 1'd1;
        end else begin
          exp0_go = 1'd0;
        end
    end
    final begin
      $display("[mem]: %d", Res0_0.mem);
    end
endmodule
```

wsnyder, 2020-12-02:
set top->go = 0 at time zero?

rachitnigam, 2020-12-02:
No luck. However, adding an `initial` block and zeroing out all the values works. I think that solution is workable for me but I'm still curious to understand if there is anything going on with the extra write changing anything.

wsnyder, 2020-12-02:
The extra write turns off some optimizations which is removing temporary variables.  Because it's a time-zero problem the state of temporaries and order of evaluations matters.  Ideally it shouldn't, but this is a tradeoff for non-time-zero speed (the common case of course).

rachitnigam, 2020-12-02:
Got it. Thanks a lot for the prompt responses and the awesome open source tooling!

wsnyder, 2025-04-25:
See #2262.

wsnyder, 2025-04-25:
See #2662