verilator/verilator #2663: Systemc Example not running

rahulraveendran15-coder, 2020-12-02:
How may we help - what is your question?
I have tried using systemc with verilator and it complains the below error:
[image]

I am using systemc version 2.3.3 from accelera website. My system is centos 7 64 bit

wsnyder, 2020-12-04:
This is likely not directly related to Verilator, something looks wrong in your SystemC installation, try SystemC's examples, and look at if the headers match the link libraries being used.  Report back if you could to help who gets it next.  (BTW nicer to put in the text of the error rather than a picture so it's searchable and easier to read.)

wsnyder, 2020-12-06:
Feel free to reopen if it is really something Verilator related, and also please do report back what was the problem if you can, thanks.

closed 2020-12-06 by wsnyder