verilator/verilator #2664: Design Ware

simdream, 2020-12-04:
If the design contains DW, is there any good solution? DW library is encrypted either with VCS or IUS so seems like a need of open source DW ?

wsnyder, 2020-12-04:
There is yet no good solution for design ware.  Certainly would be good to start an open-source project to hand-code (cleanroom-wise) an open-source design-ware replacement for the smaller parts (data path etc).  Behaviorally (vs synthesis) should be much easier.  Perhaps someone has started this but I'm unaware.

An alternative someday maybe a --protect-lib or similar library from the vendor, but not ready nor in enough demand yet.

closed 2020-12-04 by wsnyder

simdream, 2020-12-04:
--protect-lib generated the scheduled code for the IP itself. With many DW instances big and small, I assume too many eval()s happening, not mentioning there might be potential order issue.

DW is very parameterized, to synthesis to gate does not provide the flexibility.

Will IEEE encryption be an option and in roadmap? My point is, even verilator is open source, but the design IP and VIP/BFM are still protected, which could be a great blocking issue for adoption of verilator.

wsnyder, 2020-12-04:
I'm not sure how IEEE protection would work with an open source tool as a user could simply modify the tool to get around it, but am open to ideas.  This was part of the idea of protect_lib in that the vendor would run verilator so can ensure integrity.

We probably can figure out some solution to allow parameters and make protect_lib reasonably performant.  But a mix of multiple approaches might be more ideal, an open source pin/name-compatible DWC for small stuff (this is same as an API so as long as designed without looking into DWC sources would be legal), then protect lib for big IP like USB controllers.